pycom/pycom-micropython-sigfox #302: Build not working

syndycat, 2019-05-17:
Board: GPP,LOPY 

**make BOARD=GPY TARGET=app**
Use make SECURE=on [optionally SECURE_KEY ?= secure_boot_signing_key.pem] to enable Secure Boot and Flash Encryption mechanisms.
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: *** No rule to make target 'build/GPY/release/frozen_mpy/frozen/Base/_boot.mpy', needed by 'build/GPY/release/frozen_mpy.c'.  Stop.

robert-hh, 2019-05-17:
You do not need the "TARGET=app" any more. Just use:
make BOARD=GPY
Boteh bootloader and app are built then, if needed.

syndycat, 2019-05-17:
I tried again... pull from GIT 
and executing  without TARGET

same issue

Use make SECURE=on [optionally SECURE_KEY ?= secure_boot_signing_key.pem] to enable Secure Boot and Flash Encryption mechanisms.
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: *** No rule to make target 'build/GPY/release/frozen_mpy/frozen/Base/_boot.mpy', needed by 'build/GPY/release/frozen_mpy.c'.  Stop.

robert-hh, 2019-05-17:
It's missing the compiled _boot.py, which should reside in frozen/Base. 
You may also have to manually build mpy-cross first, but not having done that should create a different error.

syndycat, 2019-05-17:
I followed the steps described by you here in GIT.  

1. mpy_cross tool was builded as before step and is working.
2. _boot.py is inside esp32/frozen/Base  
3. I checked the esp32\build\GPY\release\    and seems that the folder  frozen_mpy  is not there. 

So definitely the build doesn't compile the mpy files.

robert-hh, 2019-05-17:
I have to check later my set-up, which works. I may still have copies of _boot.py and _main.py in frozen, because that is the place where they used to be.

syndycat, 2019-05-17:
I checked also running manually      mpy_cross.exe   ../esp32/frozen/Base/_boot.py 
and is working. 

So definitely there is build issue.

robert-hh, 2019-05-17:
I have no _boot.py or _mani.py in frozen. They reside in frozen/Base. How if you execute:
make clean
make BOARD=GPY
or just build for  another board, like WIPY?

Xykon, 2019-05-17:
This is a Windows specific issue and we don't officially provide support for building the firmware under Windows. Our suggestion is to build the firmware on Ubuntu Linux which can run in a VM or docker container under Windows. I'm looking into the build issue now but cannot give an ETA for a fix at this time.

syndycat, 2019-05-18:
Right. We using msys for building the esp32 binary files. As you know MSYS is an official environment dev for ESP32.  
However,  everything was running fine with pycom builds until this release.

syndycat, 2019-05-18:
> I have no _boot.py or _mani.py in frozen. They reside in frozen/Base. How if you execute:
> make clean
> make BOARD=GPY
> or just build for another board, like WIPY?

closed 2019-05-18 by syndycat

robert-hh, 2019-05-18:
The mpy file are created by the file /pycom-micropython-sigfox/py/mkrules.mk, at about line 170. Maybe it's a timestamp or path or make issue.