pycom/pycom-micropython-sigfox #303: LTE broken in 1.20.0.rc11 - regression from 1.20.0.rc10

stanleypa, 2019-05-21:
Please include the following information when submitting a bug report:

G01
Pycom MicroPython 1.20.0.rc11

LTE no longer works. I can no longer even get the ccid (see code below) and no longer get a connection.

This worked with rc10 and I tried to go back to that but it doesn't exist in 

https://software.pycom.io/downloads/GPy.html

Why not?

I went back to rc8 and the ccid code below and connection worked again.

        self.lte            = LTE()
        self.thread_at_cmd('AT^RESET')
        LTE.reconnect_uart()
        self.lte.init()
        for i in range(3):
            self.ccid=self.lte.iccid()
            print ("ccid = " + str(self.ccid))
            if self.ccid:
               break
            time.sleep(1)

iwahdan88, 2019-05-21:
Hello @stanleypa , you don't need `LTE.reconnect_uart()` or `self.lte.init()` after `self.lte = LTE()`, is there a reason behind that?

stanleypa, 2019-05-21:
when I was trying to get this working initially I came across

https://forum.pycom.io/topic/4689/read-iccid-not-working

and once it worked I never revisited this code

Are you saying the problem will go away if I remove these?

stanleypa, 2019-05-21:
FWIW:
Just checked with rc8 - if I don't have these I get the following output
ccid = None
ccid = ",""

OK

+SQNCC

With these the the iccid method always works the 2 time:

ccid = None
ccid = 89882390xxxxxxxxxxxxx

stanleypa, 2019-05-22:
Is there anywhere I can get rc10?

iwahdan88, 2019-05-22:
@stanleypa You can get it/flash it via Firmware Updater tool. or if you prefer to build the sources checkout the v1.20.0.rc10 tag

stanleypa, 2019-05-22:
if I use the firmware updater tool I get the latest (rc11) right? - which is broken. Most of the previous binaries for downgrading are stored here:

https://software.pycom.io/downloads/GPy.html

why not rc9 and rc10?

iwahdan88, 2019-05-22:
@stanleypa when using the Firmware Updater you can choose which firmware you want to flash from a drop down menu 
For the FWs under https://software.pycom.io/downloads/GPy.html.... they are available now.

iwahdan88, 2019-05-22:
However I don't see a problem in rc11, if you update your code to 
```
    self.lte            = LTE()
    for i in range(3):
        self.ccid=self.lte.iccid()
        print ("ccid = " + str(self.ccid))
        if self.ccid:
           break
        time.sleep(1)
```
you will get the same result .

The first attempt resulting in `ccid=None` is because the modem is not ready yet with iccid number thus returning response `ERROR`

if you update code to:
```
    self.lte            = LTE()
    time.sleep(3)
    for i in range(3):
        self.ccid=self.lte.iccid()
        print ("ccid = " + str(self.ccid))
        if self.ccid:
           break
        time.sleep(1)
```

You will get ICCID from first time in both firmwares.

stanleypa, 2019-05-25:
It looks like I had a H/W issue - I have tried a few other systems with rc11 and teh later sequans firmware and was able to read the ccid and connect

stanleypa, 2019-05-27:
have multiple G01 and Fipy working with this pycom and modem firmware now. I can only assume the board with the issue is H/W related

closed 2019-05-27 by stanleypa