pycom/pycom-micropython-sigfox #314: Lopy4 unable to start WLAN.AP - Guru Meditation error: core 0 panic'ed (instrfetchprohibited)

SalomonHenao, 2019-06-25:
Hello everyone,

I'm using a Lopy4 board.
Firmware:
(sysname='LoPy4', nodename='LoPy4', release='1.18.2.r7', version='v1.8.6-849-df9f237 on 2019-05-14', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

I have more than 8 Lopy4 with the same problem.
The boards are getting damaged, and I can't realize the reason yet.

1. The problem is that when trying to start the WLAN Access Point the board crashes and prints the "Guru Meditation error: core 0 panic'ed (instrfetchprohibited)" error.
2. I thought that it was because of any wrong line on my code, so I deleted everything from the boards and even updpated the pycom firmware to the last version in order to fix them, but the boards are still crashing.
3. Lopy4 launches the WLAN.AP by default on boot, so with this problem the boards are unusefull because they fall in to a loop in which they reboot all the time.
4. In order to recover the boards, I had to become creative to achieve running "wlan_on_boot(False)" before they reboot, so they quit crashing because they are not launching the WLAN.AP on boot.
5. It's difficult to diagnose because some times the rebooting happens inmediatelly, while in others it takes some time to crash... WITHOUT MOVING ANYTHING OR RUNNING ANY LINE!

Any idea?
Hope you do.
Thanks!

iwahdan88, 2019-06-25:
Hello , can you send the crash log for investigation?

SalomonHenao, 2019-06-25:
> Hello , can you send the crash log for investigation?

Sure!

Here... This is the log of a Lopy4 which is empty.
I've only run the "wifi_on_boot(True)" command and rebooted manually in order to get it:

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060430  A0      : 0x80118f25  A1      : 0x3ffbd650
A2      : 0x00000000  A3      : 0x00000037  A4      : 0xffffffb7  A5      : 0x3fbc0af4
A6      : 0x00000008  A7      : 0x00060023  A8      : 0x801241f8  A9      : 0x3ffbd640
A10     : 0x3fbc0b10  A11     : 0x00000037  A12     : 0xffffffb7  A13     : 0x3ffbd6e4
A14     : 0x7777e18c  A15     : 0x7777e18c  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4009b67d  LEND    : 0x4009b6b1  LCOUNT  : 0xffffffff

Backtrace: 0x00000000:0x3ffbd650 0x40118f22:0x3ffbd670 0x40119f2f:0x3ffbd740 0x401304c5:0x3ffbd760 0x40132027:0x3ffbd7b0

================= CORE DUMP START =================
hDcAABAAAABsAQAA
6Dv9P2A6/T/gO/0/
YDr9P4A7/T9MmHwc+FL8P/hS/D/oO/0/8FL8PwEAAABgbv0/YG79P+g7/T8AAAAA
GAAAAOQ3/T9pcGMwAImsPETV+nG2850AAAAAAOA7/T8AAAAAIQAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADMav0/NGv9P5xr/T8AAAAAAAAAAAEAAAAAAAAA
4mZAPwAAAAC4vQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAANp5yw==
pDQIQKi9CEAwBAYA3B4IgCA7/T80bv0/AAAAAIBu/T8AAAAANG79PwAAAACovQiA
ADv9PwAAAAD/////BFP8PwAAAAAEGfw/FwAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAJ02CEAAAAAA1NsIQAxk+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAOg7/T8BAAAA/gAAALgp/D8AAAAAYDv9P+hP/D8AAAAA
AAAAAAsFAAAAAAAAFwAAAP////8AAAAA6E/8P9hP/D8obv0/AAAAAAEAAAAAAAAA
AAAAAIA7/T8AAAAAAAAAAIhfCEAAAAAAAAAAAAAAAAClEwiAYDz+P3RQ/D98UPw/
AAAAAAAAAACMO/0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
WNj7P5DV+z9Q2Ps/
gNb7P/DX+z+xBq475FL8P+RS/D9Y2Ps/3FL8PwIAAADAUPw/wFD8P1jY+z8AAAAA
FwAAAFTI+z93aWZpAD6oJIORr8pWvREAAAAAAFDY+z8AAAAAIAIGABcAAAABAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADMav0/NGv9P5xr/T8AAAAAAAAAAAEAAAAAAAAA
4mZAPwAAAAC4vQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAS/PA==
BQAAAAAAAAAwBAYAJY8RgFDW+z8AAAAANwAAALf////0Crw/CAAAACMABgD4QRKA
QNb7PxALvD83AAAAt////+TW+z+M4Xd3jOF3dxAAAAAUAAAAAAAAAH22CUCxtglA
/////7f////k1vs/SDQIQHwA+T8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAACD1AwD/AAIBMAoGAAa3EoAynxGAcNb7PwEAAABICrw/
AAAAAART/D8wBgYACcwSgMgEE4BA1/s/SAq8PxcAAAAFAAAAAAAAAP3///8AAAAA
pDQIQE32EkAwAgYAvP0SgEDX+z9ICrw/9Aq8P8QKvD8BAAAA/gAAAAA3RgAMAAAA
8P///7gp/D8AAAAAWNj7PwEAAAD+AAAAAAAAABkAAAAAGAAAAQAAANi1CUAGtjCu
pOyM4Z02CEABAAAA1NsIQDsAAAADAAAAAAAAABcAAAD//z+zAAAAAKgAAAAAAAAA
AAAAAAEAAAD0Crw/AAAAAKAAAAAQC7w//wAAAGDI/D8qIBOAYNf7PwEAAAAcnxFA
AAAAABALvD8QC7w/ADdGAAAAAACw1/s/ONH8P/iO/D8AAAAAuJr8PzjR/D/4jvw/
AAAAAAAAAAA40fw/J+8GAP////+w1/s/ONH8P/iO/D880fw/9Aq8PwAAACAEAAAA
AAAAAPDX+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAARAAAAAAAAAAAAAAAAAAAA
GgAAACcgE0CEukE/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPzX+z8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
jFT8P1Cs/T9csv0/
UKz9P/Cx/T9sBAAAfFH8P3xR/D+MVPw/dFH8PxQAAAAIbv0/CG79P4xU/D8AAAAA
BQAAAGCC/T9NaWNyb1B5AAAAAAAAAAAAAQAAAFyy/T8AAAAAIAcGAAUAAAADAAAA
AAAAAIxg/D8AAAAAAAAAAAAAAADMav0/NGv9P5xr/T8AAAAAAAAAAAEAAAAAAAAA
4mZAPwAAAAC4vQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
[... 250 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAA==
7M39P0Dl/T9Y5/0/
QOX9P/Dm/T/9UWNLZEH9P5RQ/D/szf0/jFD8Pw4AAAAcyP0/HMj9P+zN/T8UyP0/
CwAAAFzP/T9JUlFzAHDnGEu3upiy6NYAAQAAAFjn/T8AAAAAIQAGAAsAAAAAAAAA
AAAAAIzm/T8AAAAAAAAAAAAAAADMav0/NGv9P5xr/T8AAAAAAAAAAAEAAAAAAAAA
4mZAPwAAAAC4vQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAtZK+w==
pDQIQKi9CEAwAwYAN4wOgADm/T/wx/0/oOb9PzzI/T8AAAAARMj9PwAAAACovQiA
4OX9PwEAAAD/////CFP8PwEAAAAEGfw/AAAAAAAAAAB8D/s/AAAAAAAAAAAAAAAA
AAAAAJ02CEABAAAA1NsIQHwP+z8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAyAAAApaWlpaWlpaU41w2AQOb9P1hf/D+g5v0/
AAAAAD0AAAClpaWlpaWlpf////9A5v0/NGL8P6Dm/T8AAAAAAQAAAExf/D8oNPw/
AAAAANDm/T9MX/w/AAAAAPDm/T8AAAAAAAAAAAAAAAAAAAAATF/8PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADQYPw/
0GD8PwAAAACg5v0/ABQAAHwP+z8AAAAAAAAAAAAAAADQYPw/AAAAAAoAAAD///9/
AAAAAPDm/T8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAD85v0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAA=
================= CORE DUMP END =================
Rebooting...

johandse, 2019-09-11:
This issue have been discussed in the forum multiple times:
https://forum.pycom.io/topic/4342/guru-meditation-error-magnetic-interference-emi-rfi
https://forum.pycom.io/topic/3892/my-new-gpy-keeps-core-dumping
https://forum.pycom.io/topic/3217/lopy4-guru-meditation-error
https://forum.pycom.io/topic/5177/frequency-disturbance-causing-my-pycom-modules-to-crash

This happens when in reach of some WIFI access points.

SalomonHenao, 2019-09-11:
> This issue have been discussed in the forum multiple times:
> https://forum.pycom.io/topic/4342/guru-meditation-error-magnetic-interference-emi-rfi [...]

Hi Johandse, yhank you for your answer.
According to that, there is nothing to do in order to solve this issue?

johandse, 2019-09-11:
Not that I am aware of. I "solved" the problem by disabling WIFI becuase I didn´t need that for my application.

amotl, 2019-11-13:
Dear @SalomonHenao and @johandse,

you might want to try our custom build just released on [1]. More background about this is available through [2].

With kind regards,
Andreas.

[1] https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/LoPy4-1.20.1.r1-0.6.0-vanilla-dragonfly.tar.gz
[2] https://community.hiveeyes.org/t/investigating-random-core-panics-on-pycom-esp32-devices/2480

amotl, 2020-02-19:
Hi there,

several people have been able to mitigate similar issues with one of our custom/unofficial Dragonfly builds based on Pycom’s 1.20.1.r1 [1] as mentioned above. We've refreshed this the other day and published corresponding Squirrel builds based on Pycom’s most recent 1.20.2.rc3 [2].

If you are lucky to try this build, we will be happy to hear about the outcome for you.
- [LoPy4-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz](https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/LoPy4-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz)

Please read both the [installation instructions](https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960#installation-instructions) and the [notes for LoPy, LoPy4, SiPy and FiPy devices](https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960#notes-for-lopy-lopy-sipy-and-fipy-devices) carefully, a) regarding the `erase_all` procedure and b) for restoring the Lora MAC after the `erase_all` procedure.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/dragonfly-firmware-for-pycom-esp32/2746
[2] https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960