pycom/pycom-micropython-sigfox #315: Socket timeout not honored with SSL

capcarr, 2019-07-02:
System information:
sysname='GPy'
nodename='GPy'
release='1.18.2.r7'
version='v1.8.6-849-df9f237 on 2019-05-14'
machine='GPy with ESP32'

When configuring a socket to use a timeout, if the socket has been wrapped with the `ussl.wrap_socket` method, the timeout is not honored.

Given a socket `sock` which has been successfully connected to a host, trying to call `sock.read` should return `None` if the timeout is reached and there is nothing to read in the socket:

```python
import usocket

sock = usocket.socket()
sock.settimeout(5)

sock.connect((host, port))

t = sock.read(1)

# after timeout, t is None
print(t is None)
```

But when the socket is wrapped with `ussl`, the timeout setting is ignored and the `read` blocks forever:

```python
import usocket
import ussl

sock = usocket.socket()
sock = ussl.wrap_socket(sock)
sock.settimeout(5)

sock.connect((host, port))

# this hangs indefinitely
t = sock.read(1)
```

This also happens at least for the `readline` method.

iwahdan88, 2019-07-18:
Hello @capcarr, probably your code hangs while trying to do `sock.connect` as this is where the handshake operation is done, so I advice when warping the socket you use the timeout option to set timeout for the handshake operation. `ussl.wrap_socket(sock, timeout=<seconds>)`
see https://docs.pycom.io/firmwareapi/micropython/ussl/

capcarr, 2019-07-18:
@iwahdan88 Thank you for your reply.
Our tests show that the socket does connect, it is the `read` method that hangs. Does the `timeout` argument in the `wrap_socket` method affect the `read` and `write` methods on the socket, or just the SSL handshake?

iwahdan88, 2019-07-18:
@capcarr No it affects only the Handshake, I tried your example code and I can see the problem. will update you with fix Asap.

iwahdan88, 2019-08-21:
Issue fixed in [2a9f77](https://github.com/pycom/pycom-micropython-sigfox/commit/2a9f771fb2fb5a18b15d313f286cca00827fe16c)