pycom/pycom-micropython-sigfox #316: Lopy4 module firmware update

phusy012, 2019-07-09:
Hi, I recently bought the lopy4. I could update the firmware for Expansion board 3.0 and now I mounted the lopy4 on the expansion board and tried to update the module firmware using the firmware tool version pycom_firmware_update_1.15.1.

However, I am not able to since I am getting the message "Failed to connect to ESP32. Time out waiting for packet header" which is as shown in the attached screenshot. If anyone can advise or help me please, 
```

Thank you!
![Capture](https://user-images.githubusercontent.com/52690767/60862679-1b3b7980-a240-11e9-9c31-be2edc35ecd4.JPG)

robert-hh, 2019-07-09:
Are you shure that you have a valid com port for the device? You can check so in the device manager of windows. And there is some trouble with the expansion board 3.1. So please check if not by chance that is yours. Expansion board 3.1 has two push buttons (USR and Safe Boot), whereas 3.0 has only one. And there should be a revision number on the board.
https://forum.pycom.io/topic/4911/expansion-board-3-1-issues

phusy012, 2019-07-10:
Thank you for the response. I wrote to support@pycom.io

Meanwhile, I checked the device manager and it does not appear in the Com port rather appears under the Libusbk USB devices. Moreover, now when I plug in the expansion board mounted with lopy4 to the computer, it gets connected, disconnected and connected again and repeats continuously. Don't know if the cable connection is loose or something is wrong. Bit confused. Please advise. Thank you

The screenshot of device manager when I plug in the device in a computer through the USB cable is shown below:
[image]

phusy012, 2019-07-11:
I realized that my laptop does not have the serial com port. Is there any method that I can use to update the firmware in lopy4?

robert-hh, 2019-07-11:
A PC serial port would not be of much use either. Normally, USB/UART bridges are used, which will be emulated as a serial port on the PC. The expansion boards have such bridges built-in.