pycom/pycom-micropython-sigfox #317: OSError when executing wlan.send_raw()

ovaag, 2019-07-09:
Board: Fipy
Basic connection: Pytrack
Firmware version: (sysname='FiPy', nodename='FiPy', release='1.20.0.rc11', version='v1.9.4-0a38f88 on 2019-05-14', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1')

Hi! I am trying to use wlan for sending raw data with the function wlan.send_raw(). I've already updated the firmware in order to get access to the newly implemented promiscuous mode.

The code I'm running can be seen below

`import network
import time

wlan = network.WLAN(mode=network.WLAN.STA)

wlan.promiscuous(True)

wlan.channel(1)

wlan.send_raw(Buffer=bytes(64),interface=network.WLAN.STA)`

Returns this fault message:

`Traceback (most recent call last):
  File "main.py", line 10, in <module>
OSError: the requested operation is not possible`

I tried looking into the release candidate branch( https://github.com/pycom/pycom-micropython-sigfox/blob/release-candidate/esp32/mods/modwlan.c#L1999 ) and found that before the data is sent, a function esp_wifi_80211_tx(ifx, value_bufinfo.buf, value_bufinfo.len, args[2].u_bool)  generates my fault message, but I was not able to get to the bottom of the problem.

Thank you!

iwahdan88, 2019-07-09:
Hi @ovaag , the `wlan.send_raw` should be given a valid raw raw ieee80211 frame.
Currently only support for sending beacon/probe request/probe response/action and non-QoS data frame

ovaag, 2019-07-11:
Thanks, it worked out! Do you have any tips for a good callback routine on the receiver side of these wlan.send_raw() packets?

closed 2019-07-11 by ovaag

iwahdan88, 2019-07-11:
@ovaag , I think you mean a use case to capture wifi packets (promiscuous mode), if so there is an example here on how to use that: https://development.pycom.io/firmwareapi/pycom/network/wlan/

elmaghba, 2022-09-10:
Could you please show me how you used it to send Beacon and request/probe frames?

Thank you!