pycom/pycom-micropython-sigfox #318: G01 I2C bus not finding I2C device (SSD1306)

mamelmahdy, 2019-07-12:
Board information:
(sysname='GPy', nodename='GPy', release='1.18.2.r7', version='v1.8.6-849-df9f237 on 2019-05-14', machine='GPy with ESP32')

I am actually using the G01 module on a custom board. I thought I could fix the problem with a firmware update, and I chose GPy as the board.

Trying to use a SSD1306 screen over the I2C bus. The same script works on a GPy connected to a PyTrack.

For initializing the screen I have

from machine import I2C
import ssd1306

i2c = I2C(0, I2C.MASTER, baudrate=100000, pins=("P9","P10"))
oled = ssd1306.SSD1306_I2C(128, 32, i2c)

Which results in this error:

robert-hh, 2019-07-12:
Please check the connection. G01 and GPy should not differ with respect to I2C on P9 and P10.

mamelmahdy, 2019-07-17:
Thank you for the response! I will revisit this as soon as I can resolve some network attachment and SIM card read issues I am having on the GPy modem (and not my custom board).