pycom/pycom-micropython-sigfox #319: Pycom Expansion Board 3.1 Firmware Update and its connection to PC

phusy012, 2019-07-13:
I procured the Pycom Expansion Board 3.1 and the Lopy4 to function as a Lora module recently.

First while trying to update the Expansion Board 3.1 firmware, did the following:
- Connected the Expansion board with USB cable (not sure if it is the right one) and followed steps using the documentation and video tutorial https://core-electronics.com.au/tutorials/pycom-pysense-pytrack-firmware-update.html

- However, I am not able to see the active Com Port as shown in the video although I am able to see Libusbk USB device (attached),

- Once the expansion board firmware is updated as same as the procedure in the video, then I unplugged and plugged in the board again.

- But now, when I plugged in the board again to the PC, the device gets connected, then in a few second the device gets disconnected, then get connected and then disconnected again and again. The sound is audible, again and again........It can be seen in device manager where it blinks again and again.

- I also tried to mount the Lopy4 on the expansion board and tried to update its firmware tool version pycom_firmware_update_1.15.1. but it gives me the message "Failed to connect to ESP32. Time out waiting for packet header".

- I read many forums and feels that serial connection is required (not sure), but I am of the view that my laptop has no serial COM port(not sure).

Please advice me, how and what should I require and any help that could help me to correctly update the expansion board 3.1 firmware first and then the Lopy4 module firmware update later. May help me both in hardware and software details that are needed ( I lack basics in fact).

Thank you

[image]

robert-hh, 2019-07-13:
You may have been hit by this bug: https://forum.pycom.io/topic/4911/expansion-board-3-1-issues
It causes the expansion board getting unresponsive after software update.

phusy012, 2019-07-14:
Thank you @robert-hh 
I contacted with the Pycom support but did not get the confirmation yet. Can another option be buying a new expansion board like 3.0 instead of 3.1? Or a Pysense/Pytrack board can also serve the function to mount Lopy4? Bit worried due to the project deadline.

Actually, we will be using the Lora node using the Lopy4 and will be connected to the Gateway Rak831 Pilot Gateway. Any suggestion or help will be highly appreciated. Thanks

robert-hh, 2019-07-14:
You can use an expansion board 3.0, 2.0 or Pysense/Pytrack/Pyscan. Actually do do not even need any of these boards. You can also use any USB/UART adapter with a  little bit more inconvenience. Best put both LoPy4 and USB/UART adapter on a breadboard, connect GND->GND(Lopy4), 5V -> Vin(LoPy, Tx->Rx(lopy4) via a 4.7k resistor, RX-> TX(LoPy4) (no resistor needed).
For firmware update of the LoPy4 you have to connect P2(Lopy4) to GND and push reset, then start the update. Sometimes, a lower update speed is needed.

phusy012, 2019-07-14:
Thank you @robert-hh. Will try that too. Thank you so much.

phusy012, 2019-07-24:
Hi  @robert-hh  It has been a long time and I received my USB/UART order only today.

I also referred to the https://alepycom.gitbooks.io/pycom-documentation/content/chapter/gettingstarted/connection/lopy4.html#diy where a button is connected between a pin P2 and GND. If we don't have the button connected (P2 and GND directly connected), are we suppose to press the RESET black button on the Lopy4 in order to make the Lopy4 into bootloader mode to update Lopy4 firmware? 

Please don't mind as I may keep bothering you, again and again, seeking your advice.

Thank you

robert-hh, 2019-07-24:
Yes. After connecting P2 and GND, you have to push the reset button. The procedure follows what is documented for expansion board 2: https://docs.pycom.io/gettingstarted/installation/firmwaretool/ (G23 is P2)

phusy012, 2019-07-24:
Thank you @robert-hh

phusy012, 2019-07-25:
I tried connecting the Lopy4 pins using the jumper wires and the USA/UART but still, I am not able to see the flashing blue light from Lopy4. I doubt the connections may be loose while using the jumper wires on a breadboard or the voltage too low to turn ON the Lopy? Any suggestion?

Meanwhile, on a safer side, I also ordered the Pysense, Pyscan and Pytrack which I already received. I also tried to update the firmware for the Pysense but when I connect it using the USB 2.0 cable, I can see Pysense appearing under the "Other Device" in my Device Manager but I am not able to see under the Ports (COM & LPT) (screenshot attached below.

However, when I connect the USB/UART device to the computer using the USB 2.0 cable, the USB/UART device is seen under the Ports (COM&LPT) as USB Serial Port (Com6).

Why don't we see the Pysense under the Port (COM & LPT) in Device Manager? Any driver software that we require to install? Some link is given in Pycom site https://docs.pycom.io/gettingstarted/installation/drivers/  but could not get the file. Any help please.
[image]
[image]

robert-hh, 2019-07-25:
The test with USB/UART failed most likely due to power issues. The 3.3V output of the devices can typically not supply sufficient power. Even if, you must ust Vin of LoPy4 for power supply. If ouy adapter has a 5V output, use that one. It is usually directly connected to Vusb.
On Windows, the new Pycom expansion boards need a specific driver. There are many threads in the forum about that. I cannot tell, because I do not use Windows for dealing with MicroPython.

phusy012, 2019-07-26:
Hi. Finally I could power ON the Lopy4 on a breadboard using the USB/UART. And I followed the procedure shown in https://learn.sparkfun.com/tutorials/how-to-load-micropython-on-a-microcontroller-board/pycom-lopy4 using the firmware pycom_firmware_update_1.15.1 (1).exe from https://docs.pycom.io/gettingstarted/installation/firmwaretool/.

Followed the procedure in running the firmware tool but again it shows "Failed to connect to ESP32: Invalid head of packet (\x08)".

I can see the LED in USB/UART blinking though when I tried to update the firmware on Lopy4 but not able to do it. Any suggestion please.

Thank you

robert-hh, 2019-07-26:
> Followed the procedure in running the firmware tool but again it shows "Failed to connect to ESP32: Invalid head of packet (\x08)".

Try to lower the baud rate. Some of the USB/UART bridges cannot cope with high baud rates. And theer is another updater at https://forum.pycom.io/topic/3425/new-beta-firmware-updater-1-15-2-b0.

phusy012, 2019-07-29:
Now I could connect update the Lopy4 firmware. Thanks.
Also could install Atom for PyMakr plugin and could communicate with the Lopy4. Tried changing the LED colour of the Lopy4 and it worked.

Actually I am new to the Micro Python. My initial task is to send the packets from Lopy4 to the Gateway RAK831 Pilot Gateway and the Gateway should receive the packet, thus forming the network between the node and gateway. Any suggestions will be appreciated.
Thank you