pycom/pycom-micropython-sigfox #323: How can I log any errors encountered in the MicroWebSrv server

jdcrunchman, 2019-07-23:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
I'm using the ESP32 board.

* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
import os
os.uname()
```
(sysname='esp32', nodename='esp32', release='1.11.0', version='v1.11-45-g14cf91f70 on 2019-06-16', machine='ESP32 module with ESP32')
```
* Exact steps to cause this issue
    1. First I did this...

I activated the WLAN and Hot Spot.
## Set up the Access point
ap = network.WLAN(network.AP_IF)
ap.ifconfig(('192.168.84.1', '255.255.255.0', '192.168.84.1', '192.168.84.1')) # ip, netmask, gateway, dns
ap.active(True)
ap.config(essid='aWhisper', authmode=network.AUTH_WPA_WPA2_PSK, password='')

    2. Then, I did that...
I ran the original main.py code from the github
then I went to my laptop,  and connected to the WIFI,  then I checked network connectivity
by sending a test ping.   The Ping worked just fine.

Then,  using the browser on the Mac,  and pointed it to:  http://192.168.84.1/index.html
The startup screen came up just fine.

But when I went to another page,  I get a 405 error.
Is there anyway I can trace through the server code and turn on some server logging, so I can
try and debug why this doesn't work?

I'm sending the full ZIP of all the HTML, IMG, CSS files
[WhisperMagicMessagingUI.zip](https://github.com/pycom/pycom-micropython-sigfox/files/3423348/WhisperMagicMessagingUI.zip)

This is my main entry code I run,  right after the boot.py.

mainSetup.py - I run this right after the boot.py
-----------------------------------------------
import network

### Setup the WLAN
wlan = network.WLAN(network.STA_IF) # create station interface
wlan.active(True)       # activate the interface
myscan = wlan.scan()             # scan for access points
wlan.isconnected()      # check if the station is connected to an AP

print ("WLAN Setup")

### Set up the Access point
ap = network.WLAN(network.AP_IF)
ap.ifconfig(('192.168.84.1', '255.255.255.0', '192.168.84.1', '192.168.84.1')) # ip, netmask, gateway, dns
ap.active(True)
ap.config(essid='aWhisper', authmode=network.AUTH_WPA_WPA2_PSK, password='my-password')

print ("Access point setup - IP = " + "192.168.84.1")

print ("creating an array of table values")
mytable = [['john', "not listening", "36db"], ['andy', 'Online', '56db']]

print ("Now run mainweb.py to start the server")
import mainweb

Then run "mainweb.py" below
--------------------------------
from microWebSrv import MicroWebSrv
from trace import bp

#bp('testing', (1,2,3,5)[3], 3**5)

# ----------------------------------------------------------------------------

@MicroWebSrv.route('/test')
def _httpHandlerTestGet(httpClient, httpResponse) :
    content = """\
    <!DOCTYPE html>
    <html lang=en>
        <head>
            <meta charset="UTF-8" />
            <title>TEST GET</title>
        </head>
        <body>
            <h1>TEST GET</h1>
            Client IP address = %s
            <br />
            <form action="/test" method="post" accept-charset="ISO-8859-1">
                First name: <input type="text" name="firstname"><br />
                Last name: <input type="text" name="lastname"><br />
                <input type="submit" value="Submit">
            </form>
        </body>
    </html>
    """ % httpClient.GetIPAddr()
    httpResponse.WriteResponseOk( headers        = None,
                                  contentType    = "text/html",
                                  contentCharset = "UTF-8",
                                  content        = content )

@MicroWebSrv.route('/test', 'POST')
def _httpHandlerTestPost(httpClient, httpResponse) :
    formData  = httpClient.ReadRequestPostedFormData()
    firstname = formData["firstname"]
    lastname  = formData["lastname"]
    content   = """\
    <!DOCTYPE html>
    <html lang=en>
        <head>
            <meta charset="UTF-8" />
            <title>TEST POST</title>
        </head>
        <body>
            <h1>TEST POST</h1>
            Firstname = %s<br />
            Lastname = %s<br />
        </body>
    </html>
    """ % ( MicroWebSrv.HTMLEscape(firstname),
            MicroWebSrv.HTMLEscape(lastname) )
    httpResponse.WriteResponseOk( headers        = None,
                                  contentType    = "text/html",
                                  contentCharset = "UTF-8",
                                  content        = content )

@MicroWebSrv.route('/edit/<index>')             # <IP>/edit/123           ->   args['index']=123
@MicroWebSrv.route('/edit/<index>/abc/<foo>')   # <IP>/edit/123/abc/bar   ->   args['index']=123  args['foo']='bar'
@MicroWebSrv.route('/edit')                     # <IP>/edit               ->   args={}
def _httpHandlerEditWithArgs(httpClient, httpResponse, args={}) :
    content = """\
    <!DOCTYPE html>
    <html lang=en>
        <head>
            <meta charset="UTF-8" />
            <title>TEST EDIT</title>
        </head>
        <body>
    """
    content += "<h1>EDIT item with {} variable arguments</h1>"\
        .format(len(args))

    if 'index' in args :
        content += "<p>index = {}</p>".format(args['index'])

    if 'foo' in args :
        content += "<p>foo = {}</p>".format(args['foo'])
#    bp('content = ', content)
    content += """
        </body>
    </html>
    """
    httpResponse.WriteResponseOk( headers        = None,
                                  contentType    = "text/html",
                                  contentCharset = "UTF-8",
                                  content        = content )

# ----------------------------------------------------------------------------

def _acceptWebSocketCallback(webSocket, httpClient) :
    print("WS ACCEPT")
    webSocket.RecvTextCallback   = _recvTextCallback
    webSocket.RecvBinaryCallback = _recvBinaryCallback
    webSocket.ClosedCallback     = _closedCallback

def _recvTextCallback(webSocket, msg) :
    print("WS RECV TEXT : %s" % msg)
    webSocket.SendText("Reply for %s" % msg)

def _recvBinaryCallback(webSocket, data) :
    print("WS RECV DATA : %s" % data)

def _closedCallback(webSocket) :
    print("WS CLOSED")

# ----------------------------------------------------------------------------

#routeHandlers = [
#   ( "/test",  "GET",  _httpHandlerTestGet ),
#   ( "/test",  "POST", _httpHandlerTestPost )
#]

srv = MicroWebSrv(webPath='/')
srv.MaxWebSocketRecvLen     = 256
srv.WebSocketThreaded       = False
srv.AcceptWebSocketCallback = _acceptWebSocketCallback
srv.Start()

There are other files in this document as well,  such as css,  images,  and such.   
Am I using some feature in my HTML that is not supported on the MicroWebSrv package?
All of the files are in the attached ZIP file.

Thank you!

robert-hh, 2019-07-23:
Wrong board. Please refer to forum.micropythpn.org

jdcrunchman, 2019-07-24:
Sorry, the website *forum.micropythpn.org
<http://forum.micropythpn.org>* cannot
be found

Can you please direct me to the proper page?

robert-hh, 2019-07-24:
https://forum.micropython.org/