pycom/pycom-micropython-sigfox #328: wlan callback is triggered multiple times for every incoming packet

ovaag, 2019-08-01:
Board: Fipy
Basic connection: UART
Firmware version: (sysname='FiPy', nodename='FiPy', release='1.20.0.rc12', version='v1.9.4-81167ed on 2019-07-15', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1')

I have a setup with two Odroid XU4 and two fipys. The pipeline for a message is the following:

""
odroid1 --> UART --> FiPy1 --> wlan --> FiPy2 --> UART -->odroid2
""

The code on FiPy2, which is receiving raw wlan messages is the following:

```                                           
import pycom
import time
from machine import UART
import utime
import _thread
from network import WLAN
from network import LoRa
import uhashlib
import socket

#LoRa settings
lora = LoRa(mode=LoRa.LORA,frequency=868100000,region=LoRa.EU868,tx_power=10,sf=7,coding_rate=LoRa.CODI$
lora.bandwidth(LoRa.BW_250KHZ)
s = socket.socket(socket.AF_LORA,socket.SOCK_RAW)

#wlan settings
wlan = WLAN(mode=WLAN.STA,auth=None,bandwidth=WLAN.HT20)
wlan.promiscuous(True)
wlan.channel(2)
wlan_msg = bytes()

nqos_frame = bytes([0x08, 0x03])        #Non-Qos data frame
wlan_msg += nqos_frame
wlan_msg += bytes(22)   #Duration and address fields not in use. Fill with zeroes

#UART settings
uart = UART(1, baudrate=115200)
uart.init(115200,bits=8)

#Set led color to red
pycom.heartbeat(False)  
pycom.rgbled(0x100000)

count = 0
c = 0
def push_uart(data,type):
    msg_len1 = bytes([len(data)>>8])
    msg_len2 = bytes([len(data)])
    stop_pad = bytes([0xf0,0xf0])
    print(data)
    print("")
    uart.write(msg_len1+msg_len2+type+data+stop_pad)
    pycom.rgbled(0x001000)

def DecodeRaw(frame):
    return {"data":frame[24:-7],"checksum":frame[-7:-4]}

def wlan_cb(wlan):
    global count  
    pk = wlan.wifi_packet()
    data = pk.data
    if wlan.events() & WLAN.EVENT_PKT_DATA:
        if (data[:2] == b'\x08\x0b'):
            msg = DecodeRaw(data)
            md5 = uhashlib.md5(msg["data"]).digest()[:3]
            if(msg["checksum"] == md5):
                push_uart(msg["data"],bytes([0x01]))
                count += 1
                print("count: ")
                print(count)

wlan.callback(trigger = WLAN.EVENT_PKT_DATA, handler=wlan_cb)
```

The problem I am having is that for every wlan packet being sent from FiPy1, FiPy2 receives multiple of the same packet. Typically if I send one packet from FiPy1, Fipy2 receives 20 identical packets. 

Seems like the trigger allows the callback to read the packet multiple times before a new packet is available. The whole system is running on 1 Hz.

husigeza, 2019-08-10:
Hello,
I have tried your script and could not reproduce the issue.
Do you experience that the "wlan_cb" is called several times ? Or do you experience that the same data arrives to odroid2 several times via UART?
Could you also please attach that part of the code where you send out the raw packets  on FiPy1 ?
Could you also please clean up the script to only contain the necessary part, e.g. no LoRa etc.

ovaag, 2019-08-13:
Hello Géza,

Attached is a cleaned up version of the script I posted on Github + the script that is being run on Fipy1 to send the raw packets.

Yes the problem is that wlan_cb is called several times for every message I receive.

If you want a quick replica of the error, just run main.py on one fipy and then send a raw wlan packet from console on another fipy. Just make sure to start your message with the non-qos data frame:  b'\x08\x03'

husigeza, 2019-08-22:
After investigating this issue it turned out that the packets considered as duplicated ones are actually retransmitted frames of the first frame, because the receiver did not receive the frame sent first.

closed 2023-02-15 by ovaag