pycom/pycom-micropython-sigfox #329: Can't change UART configuration while running MicroWebSrv

SalomonHenao, 2019-08-05:
Hello everyone,

I'm using a Lopy4 board.
Firmware:
(sysname='LoPy4', nodename='LoPy4', release='1.18.2.r7', version='v1.8.6-849-df9f237 on 2019-05-14', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

I have a problem because I need to read SDI12 sensors using an RS485 breaktrough while running an HTTP server [https://github.com/jczic/MicroWebSrv/](MicroWebSrv).

The problem is that the LOPY board reboots inmediately when I run:
`from machine import UART`
`uart = UART(1, baudrate=1200)`

Those lines work perfectly when the MicroWebSrv HTTP server is not running.
But when the script is executed from a method inside the API served by the HTTP server, the device reboots.
I don't know if the HTTP server uses the UART ports or something similar, because the SDI12 sensor reading works without any problem when the script is not executed from the HTTP server.

Also, the log of the reboot has not any logic, usually is full of wrong symbols and is not the same at every reboot.

Any idea? Thank yu for your support.

robert-hh, 2019-08-06:
Do you use the SD card in your code? The SD card uses P4, which is the RX pin for UART1.
And can you show a simple example that crashes?

SalomonHenao, 2019-08-06:
> Do you use the SD card in your code? The SD card uses P4, which is the RX pin for UART1.
> And can you show a simple example that crashes?

Hello Robert,
Thank you for your answer.

Actualy I'm not using the SD card. Let me be more clear:

In a common situation I can run the UART configuration in order to do a SDI12 communication.
The code would look like:
`def sdi12():`
   `from machine import UART`
   `from machine import Pin`
   `uart = UART(1, baudrate=1200)`
   `p_dir = Pin('P1', mode=Pin.OUT)`
   `p_tx = Pin('P3', mode=Pin.OUT)`
   `...#The code continues... and does the communication...`

Now, I start a WLAN Access Point, add the MicroWebSrv HTTP server, create a handler which runs the same script and start it:

`from network import WLAN`
`wlan = WLAN(mode=WLAN.AP, ssid="HOTSPOT", auth=(WLAN.WPA2,"12345678"), antenna=WLAN.INT_ANT)`

`import MicroWebSrv`
`handlers = [( "/api", "GET", sdi12)]`
`srv = MicroWebSrv(routeHandlers=handlers)`
`srv.Start(threaded=True)`

Here the problem appears. When I connect an HTTP Client to the device and run the GET method, the SDI12 function runs correctly but the device crashes exactly when running this line:
`uart = UART(1, baudrate=1200)`

The MicroWebSrv works perfectly with other scripts but when I try to change UART configuration through one of it's functions, the device crashes.

husigeza, 2019-08-10:
Hello,
So far I could not reproduce the crash, however I noticed one thing. Do you use an expansion board ? Because P1 is used for the serial communication between the PC and the device. 
Please use a PIN other than P1, or try running your script when you remove the "RX" jumper from the board. You can telnet to the device in this case to get the REPL and see the output. When the jumper is not removed, the device freezes for me when "p_dir = Pin('P1', mode=Pin.OUT)" is called. It may happen that the crash you notice is not during "uart = UART(1, baudrate=1200)", rather it is at the next line.

SalomonHenao, 2019-08-12:
Hello @husigeza , thank you for your support.

I've been testing and actually the crash happens when the next line is called:
`p_dir = Pin('P1', mode=Pin.OUT)`

But the LoPy4 is not attached to any dev board,  because it's actually embebed over a custom board where the USB communication pins are not connected to any data line. P1 is connected to the RS485  interface using a digital line to control de data flow direction.

I checked using P23 instead of P1, but it crashes anyway.

I don't know if it does matters, but the LoPy4 is starting the HTTP server over a thread and simultaneously starts the telnet server over another thread printing an administrator menu. When the SDI12 function is called from the Telnet server it works fine, but when is called from a method contained inside the HTTP API, it crashes.

SalomonHenao, 2019-08-12:
Hello,

I've been testing and realized that the bug was caused because of a Stack Overflow that was happening right while the USB configuration was unable to work, so the device crashed without being able to print a readable message.

When changed P1 to P23, the device crashed anyway but was able to print the error message:
`A stack overflow in task MPThread has been detected`

I've reduced the amount of levels that it was running and the bug became solved without replacing P1.

Thank you for your help!

closed 2019-08-12 by SalomonHenao