pycom/pycom-micropython-sigfox #330: SIGFOX becomes freezed while publishing

SalomonHenao, 2019-08-05:
Hello everyone,

I'm using a Lopy4 board.
Firmware:
(sysname='LoPy4', nodename='LoPy4', release='1.18.2.r7', version='v1.8.6-849-df9f237 on 2019-05-14', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

I have been a whole year developing a commercial IOT device based in LOPY4.
Some weeks ago I bougth 30 LoPy4 units and assembled 30 units of this IOT device.

When testing the new devices an issue appeared, but is an issue that appeared with this new group of LoPy4. I didn't change any line of my code but now the devices become freezed while trying to publish data to SIGFOX.

The device runs `s.send(bytes(data))` as always to publish data to SIGFOX, but becomes freezed and doesn't publish. If I reboot manually the device, it publishes sometimes and other times becomes freezed again.

I tried to downgrade the micropython firmware to older versions but it doesn't solves the problem. And if I use the same code with older LoPy4 (I have around 100 units) it works fine.

I hope that you know what is happening.

SalomonHenao, 2019-08-06:
I have been testing with a current consumption logger, and realized that it is related to the SIGFOX publication time... If I try to make publications with time less than 2 minutes it crashes.

In the next image you can see the device (trying to do a sigfox publication every 1 minute) that wakes up from deep sleep, reads data from sensors and then try to do a SIGFOX publication, but becomes freezed until that a WatchDog reboots the system, and then the SIGFOX publication becomes done. The problem repeats in every iteration:
[image]

Now, in the next image you can see a device with publication time of 3 minutes in which the problem is solved:
[image]

**The problem is that some times it happens at the first publication, not when trying to publicate very soon. I actually can cut off the power source and restore it, but the LoPy4 freezes trying to publicate... This Issue never happened with the other LoPy4 units, it's only with this group of 30 units.**

Note: I know that SIGFOX doesn't accept these publciation periods but it's only for development and testing purposes.

joelmarini98063454, 2019-10-09:
I too have been experiencing this issue, I bought units earlier in the year and they worked fine up until about a month ago. I cant remember if it started after a firmware update or not. I am always able to send 1 message however no more after that. I am in RC4, not sure if that makes a difference.

SalomonHenao, 2019-10-11:
> I too have been experiencing this issue, I bought units earlier in the year and they worked fine up until about a month ago. I cant remember if it started after a firmware update or not. I am always able to send 1 message however no more after that. I am in RC4, not sure if that makes a difference.

I have been trying to work on it by myself and now it seems to be solved.
I noticed that when I run the same lines that my code uses for Sigfox communication but out of my code in a plain script, it works.
So I worked in changing the deepness of the layer that runs the Sigfox communication inside my code and it became working. I'm not 100% sure, but I guess it was because of a memory overflow.

Anyway, it never happened with older versions. It's not very nice to have to reorder my code withot a valid explanation.

SalomonHenao, 2019-11-05:
> I too have been experiencing this issue, I bought units earlier in the year and they worked fine up until about a month ago. I cant remember if it started after a firmware update or not. I am always able to send 1 message however no more after that. I am in RC4, not sure if that makes a difference.

I am using RC4 too.
I solved this issue identifying that the problem appeared when creating a Sigfox object in a different thread that which makes the actual connection, some seconds before connecting to the network.
Now the firmware doesn't creates Sigfox objects in different threads... Only in the specific moment where it's needed to publish data.

doniks, 2020-02-05:
Hi guys. 

Please retest with the firmware release from today https://github.com/pycom/pycom-micropython-sigfox/releases/tag/v1.20.2.rc3 

It fixes an issue with a sigfox deadlock in rc2/4.