pycom/pycom-micropython-sigfox #332: Lopy4 abort() was called at PC on core 0 (stuck in boot loop) after wake up from deepsleep

soibac35, 2019-08-21:
board : LOPY 4

REPL output :

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8028,len:8
load:0x3fff8030,len:1688
load:0x4009fa00,len:0
load:0x4009fa00,len:14592
entry 0x400a059c
abort() was called at PC 0x400814cc on core 0

Backtrace: 0x4008f1a7:0x3ffe3c80 0x4008f1d8:0x3ffe3ca0 0x400814cc:0x3ffe3cc0 0x400a0548:0x3ffe3ce0 0x400a066d:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Backtrace: 0x4008e8cf:0x3ffe38c0 0x4008d7c9:0x3ffe38e0 0x4008f73b:0x3ffe3910 0x4008fa7a:0x3ffe3ae0 0x4008f434:0x3ffe3b20 0x4008f5c4:0x3ffe3ba0 0x4008349a:0x3ffe3bc0 0x4008f1a4:0x3ffe3c80 0x4008f1a4:0x3ffe3ca0 0x400814cc:0x3ffe3cc0 0x400a0548:0x3ffe3ce0 0x400a066d:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Re-entered core dump! Exception happened during core dump!
Rebooting...

ste1387, 2019-10-15:
I got kind of same error after machine.deepsleep on my LoPy4... Any workaround?

------------
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8028,len:8
load:0x3fff8030,len:1728
load:0x4009fa00,len:0
load:0x4009fa00,len:14584
entry 0x400a059c
abort() was called at PC 0x400d33e2 on core 1

Backtrace: 0x4008edb7:0x3ffdaea0 0x4008ede8:0x3ffdaec0 0x400d33e2:0x3ffdaee0 0x400e8836:0x3ffdaf00 0x400dce9c:0x3ffdaf30

================= CORE DUMP START =================
jCMAAAsAAABsAQAA
HFT8P+Ct/T/wsP0/
MK39P5Cw/T8AAAAAhFH8P4RR/D8cVPw/fFH8Pw4AAAAgbf0/IG39PxxU/D8AAAAA

soibac35, 2019-10-15:
disable brown-out in their firmware or easier way is to update to development version. That works.
I am not sure why they have that in official version and not fix it yet.

ste1387, 2019-10-22:
Thanks a lot for the hint! I've been trying to implement what you suggested but probably I didn't get your point properly. Should I follow [this pos](https://forum.pycom.io/topic/2446/building-my-own-firmware/17)t to build my own firmware without brown-out? Otherwise, how can I switch to development version?

soibac35, 2019-10-22:
You can build your own firmware but i wouldn't recommend you dont do it if you are not familiar with them. 
Instead, Use firmware update software from pycom, then there should be a field say stable version, select it and pick development. Then proceed to the end