pycom/pycom-micropython-sigfox #334: LoPy4 non default I2C pins not working

DevelopAll, 2019-08-27:
LoPy4 running 1.16.0

I am trying to connect my LoPy4 to a MAX11607 ADC chip with its peripherals via I2C. Now when using the default SDA and SCL pins (P9 and P10 respectively) the function I2C.scan() returns [52] which is correct as that is the address of the MAX11607. However, once I declare I'd like to use pins P3 and P4 I2C.scan() will only return empty arrays as it cannot find the chip. Interestingly, when I declare P3 and P4 as my I2C pins the MAX11607 is still recognised if I connect it to P9 and P10. So it seems like the change of pins did not come through. I have tried this with another I2C device too which gave the same results.

The code I am using is below:
`from machine import I2C
import time

i2c = I2C(0)
i2c = I2C(0, mode=I2C.MASTER, pins=('P3', 'P4'))
i2c.init(I2C.MASTER, baudrate=400000)

while True:
    print(i2c.scan())
    time.sleep_ms(20)`

Any idea what could cause this?

robert-hh, 2019-08-27:
I2c.init() resets the default pins. You can also set the baudrate in the constructor, or you have to specify the pins in the call to init.

DevelopAll, 2019-08-28:
Thank you Robert,
Using your solution my setup works!
I completely oversaw that init() resets to the default pins when following the I2C documentation of Pycom.

closed 2019-08-28 by DevelopAll

robert-hh, 2019-08-28:
> I completely oversaw that init() resets to the default pins when following the I2C documentation of Pycom.

You could not know. It is indeed surprising and caused by the way that I2C module is implemented. There is a common init helper function, which is called by both the constructor API and the init() API call.