pycom/pycom-micropython-sigfox #338: FiPy, LTE: Recurred error - no band 8 support

robert-hh, 2019-09-06:
With the change of firmware 1.20.0.rc12.1 to 1.20.0.rc13 I get the following error, when attaching the modem:
```
>>> from network import LTE
>>> lte = LTE()
>>> lte.attach(band=8, apn="iot.1nce.net")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: band 8 not supported by this board hardware!
```
It works fine, if I do NOT specify the band in the attach command, although the attach time increased a lot.

robert-hh, 2019-10-07:
With the new release 1.20.1 this bug still persists. In handling of the attach command inside modlte.c, the length of the command AT+SMDD is not set! Lines affected: 310, 654, 670. Compare to these place the use of the function lteppp_send_at_command in line 179, where the data length IS set.

iwahdan88, 2019-10-07:
Hello @robert-hh 
Can you open a PR on Dev branch for your fix, 
Thanks.

robert-hh, 2019-10-07:
I'll do.

robert-hh, 2019-10-07:
I did.

closed 2019-10-07 by robert-hh