pycom/pycom-micropython-sigfox #341: Memory Leak on Bytearray Assignation to Slice.

jmalcaraz, 2019-09-14:
Pycom board: FiPy 
Firmware: 1.20.0.rc11.2 - (It is suspected that ALL firmwares will be affected)

The following code (ready to execute) generate 2 ring buffers of bytearrays. 
Rings the same size (8). 
The size of the bytearrays stored in each ring are different. 
The bigger bytearray size is x32 times the smoller one (128 bytes).  
For each big bytearray,  we make 32 slices to asign the values of small bytearray. 

The code should not consume any memory since all bytearrays are created at the begining and no more creations are done. However, it consumes all the memory in seconds.  I have provided in the comments at least 3 different variations, with the usage of memoryviews. All of them produce the same problem. 

Thanks a lot in advance. 

```python
import gc
SAMPLE_BLOCK_SIZE=128
FFT_NUM_SAMPLE_BLOCK_SIZES=32
NUM_RINGS=8

rx_ring_buffer = [bytearray(SAMPLE_BLOCK_SIZE) for i in range(NUM_RINGS)]
fft_in_buffer = [bytearray(SAMPLE_BLOCK_SIZE  * FFT_NUM_SAMPLE_BLOCK_SIZES) for i in range(NUM_RINGS)]

def memoryleak():
    rx_ring_index = -1
    fft_ring_index = 0;
    current_sample = 0;
    while True:
            rx_ring_index = rx_ring_index + 1
            current_sample = current_sample +1

            # CAUSES MEMORY LEAK
            buf_view_fft = memoryview(fft_in_buffer[fft_ring_index % NUM_RINGS][(current_sample-1) * SAMPLE_BLOCK_SIZE  : (current_sample)* SAMPLE_BLOCK_SIZE ]);
            buf_view_fft = rx_ring_buffer[rx_ring_index % NUM_RINGS]

            # CAUSES MEMORY LEAK
            #fft_in_buffer[fft_ring_index % NUM_RINGS][(current_sample-1) * SAMPLE_BLOCK_SIZE  : (current_sample)* SAMPLE_BLOCK_SIZE ] = rx_ring_buffer[rx_ring_index % NUM_RINGS]

            # CAUSES MEMORY LEAK
            #buf_view_fft = memoryview(fft_in_buffer[fft_ring_index % NUM_RINGS][(current_sample-1) * SAMPLE_BLOCK_SIZE  : (current_sample)* SAMPLE_BLOCK_SIZE ]);
            #buf_view_rx = memoryview(rx_ring_buffer[rx_ring_index % NUM_RINGS])
            #buf_view_fft = buf_view_rx

            if ( current_sample % FFT_NUM_SAMPLE_BLOCK_SIZES == 0):
                fft_ring_index = fft_ring_index + 1
                print(gc.mem_free())

memoryleak()
```

robert-hh, 2019-09-14:
What you see is not a leakage but the ordinary behavior of the dynamic memory management, if you allocate and free memory objects, which obviously happens. You recover that by adding gc.collect() into the loop, e.g. before print(gc.mem_free()). That forces garbage collection at the cost of execution speed. The memory use per loop seems to be 1536/32 = 48, which is not  a lot. Some objects are created, like the memoryview object.

robert-hh, 2019-09-14:
I see another problem in your code: current_sample is not limited and grows. It should not grow beyond FFT_NUM_SAMPLE_BLOCK_SIZES.

robert-hh, 2019-09-14:
The first variant above creates two intermediate objects, a memoryview object and an intermediate slice object, which results in 32 bytes being allocated, the second only allocates the intermediate slice object of 16 bytes. Note that in any case after a while the automatic garbage collection reclaims the memory.

jmalcaraz, 2019-09-14:
@robert-hh THanks a lot for the feedback and your time for addressing this issue. You support has been very valuable. It has been definitively my lack on the behaviour of the micropython internals. I have followed your recommendations and it is now working OK. I close the issue now.

closed 2019-09-14 by jmalcaraz