pycom/pycom-micropython-sigfox #343: PyGo2 over CAT M1 - Info needed now...

secirugy, 2019-09-22:
I cannot find a single thread here on PyGo2.
Was this device scratched? Although I see it is for sale...So why nobody is using it? Buggy?
I also do not get any responses to my PyGo2 questions...strange...

I finally found a surprising information in one of online articles that claims:
_**PyGo 2 will come with a Vodafone eSIM, which according to Pycom “…will work globally on their NB IoT network. We are looking at max €3 per month for subscriptions with minimum 6 months engagement. We will also have CAT M1 with other providers such as Verizon. We are in discussions with additional carriers and expect to have others in place by launch date next year.”**_

Is it true? 
That write up was done in 2018...
What is the status of the careers in the U.S.?
Will PyGo2 work with CAT-M1 over 4G LTE of T-Mobile career? If not - what U.S. Careers will work with CAT-M1 at present time?

robert-hh, 2019-09-23:
If the quality improves, waiting is the better option. I do not like the idea of being nailed to a specific carrier. That makes the device useless for the majority of people. That may change in the future, at which time the device itself may be obsolete.

secirugy, 2019-09-23:
I am not sure what "quality" you are referring to...
I simply want to know where can I find any significant info about this device and the associated SIM it contains... and whether it works with CAT M1 in the US either with 4G LTE T-Mobile or over Hologram network with their universal SIM.
So far, I do not find anything, no tech discussion, no real documentation, no examples...