pycom/pycom-micropython-sigfox #345: LoPy4 - LORA at 433 MHz

danfourie, 2019-10-07:
Hi.

I am using a LoPy4 board and try to set up LORA using the 433 MHz band. Nowhere do I find documentation on using the 433 MHz frequency ranges?

I instantiate the LoRa radio with:

lora = LoRa(mode=LoRa.LORA, frequency=433000000)

But when doing a:

print(lora.stats().tx_frequency)

It returns 868000000 and not the intended 433000000.

What should I do to use LORA at 433000000?

robert-hh, 2019-10-07:
I made a branch based on a version of the release-candidate from Sept. 1
https://github.com/robert-hh/pycom-micropython-sigfox/tree/eu433
It may work for you at the moment. Keeping that up-to-date is a little bit tedious. Maybe Pycom will add EU433 to the list of supported settings.

robert-hh, 2019-10-07:
From the code, frequency=433000000 should work. Note that the values returned by stats() reflect the parameters form the last received packet.

robert-hh, 2019-10-07:
I just tested that. With the setting:
lora = LoRa(mode=LoRa.LORA, frequency=433000000, region=LoRa.EU868)
packets are sent. Note that you always have to start your LoPy4 from power-on if you interrupted the LoRa modem or changed the setting. Otherwise it will not work.

danfourie, 2019-10-07:
Thank you for the response. Will test again from a fresh power up. Did you test it using the normal firmware/libraries or the branched version that you posted above?

What is the procedure to upload the branched firmware to my lopy4?

robert-hh, 2019-10-07:
I used today's code. The branched version implements a EU433 region.  At the moment I do not have a binary image any more. I have built that again and made a package which you can install using the pycom updater. You can find it here: https://github.com/robert-hh/Shared-Stuff
The file name is  LoPy4-1.20.0.rc13-eu433.tar.gz (pretty obvious)

robert-hh, 2020-07-07:
The Pycom main line code now supports EU433 in the version 1.20.2.rc10 and up. Branch 'Dev' of the repository.