pycom/pycom-micropython-sigfox #349: GPy, G01

srebrnyp, 2019-10-15:
I am working with G01 and GPy LTE modem. I am trying to attach to the network however the procedure does not complete with lte.isattached == true. I am using Sequans firmware NB1-41019.dup. 
The steps I take are following:
> from network import LTE
> lte=LTE()
> lte.attach(band=20, apn="xxx")
> lte.send_at_cmd("AT+CEREG?")
'\r\n+CEREG: 3,2\r\n\r\nOK\r\n'  # looking for network
> lte.isattached()
False
> lte.send_at_cmd("AT+CEREG?")
'\r\n+CEREG: 3,1,"9FC5","011FCC67",9\r\n\r\nOK\r\n'
> lte.isattached()
False

lte.isattached() always indicates that it is not attached.

robert-hh, 2019-10-15:
Which firmware version are you using?

robert-hh, 2019-10-15:
Looking into the code, it seems to look for '+CEREG: 2,1,' and '+CEREG: 2,5,' but not for '+CEREG: 3,1,',
so it misses the confirmation. So the question is, why your card responds with +CEREG: 3,1

srebrnyp, 2019-10-15:
Hi Robert,

Thank you for the answer. Now, I could fix my problem. I set the CEREG
value manually via send_at_cmd() to 3 to get more verbose output when
registering. This is way it appears in the response. I think that you can
safely ignore the first number from CEREG response, as it is not relevant
in order to determine attachment state.

Best regards,
 Piotr Srebrny