pycom/pycom-micropython-sigfox #354: All NVS commands fail

KanyonKris, 2019-11-01:
>>> os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.20.0.rc13', version='v1.9.4-94bb382 on 2019-08-22', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

All NVS commands fail:

>>> import pycom
>>> pycom.nvs_erase_all()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: the requested operation failed
>>> pycom.nvs_get('sign', None)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Exception: Error occurred while fetching value, code: 4359
>>> pycom.nvs_set('sign', 0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Exception: Error occurred while storing value, code: 4359

robert-hh, 2019-11-01:
If you made a large firmware version change for 1.20.0, it should be required to do a full erase of the LoPy4 and reinstall all firmware and apps. There are quite a few cases like yours discussed in the forum. For full erase, either is the pycom-updater-cli program from pycom or esptool.py from espressif.
On my LoPy4, the NVS commands work. But since I do a lot of trials with the firmware, I have to erase flash from time to time anyhow.

KanyonKris, 2019-11-01:
I went from 1.18 to 1.20. I will do a full erase and report back.

Thank you for the feedback. I see your name a lot in the forums. Thanks for taking the time to help others, like me.

mc404, 2019-11-06:
I'm having the same problem after upgrading to 1.20.
The [manual](https://docs.pycom.io/advance/cli/) states that important stuff like the LPWAN MAC address would also be reset if I do a _pycom-updater-cli -p COMx erase_all_. Do I need to do a backup before erasing flash, or is there anything else I should consider before executing that command?

robert-hh, 2019-11-06:
You should backup your files or make shure, that you have otherwise copies.

mc404, 2019-11-06:
Thanks for your very quick answer! But which files do I have to backup and restore for preserving the LPWAN MAC address and other important stuff? Would the following work:

1. execute _pycom-updater-cli -p COMx copy -p factory -b_
2. execute _pycom-updater-cli -p COMx erase_all_
3. install firmware again using Pycom Firmware Update utility
4. execute _pycom-updater-cli -p COMx copy -p factory -r <backupfile.bin>_

Sorry for my naive questions, but I'm new at this and don't want to brick my LoPy4.

robert-hh, 2019-11-06:
You do not have to do anything special for the LPWAN MAC address. That will be restored when you use the Pycom Firmware Update Utility. You can use the lpwan command of pycom-updater-cli to get copies of these numbers.

pycom-update-cli -p COMx lpwan

That gives you the information for a manual restore, if the automatic one fails.

robert-hh, 2019-11-06:
Looking at the copy command, the only interesting part would be the config partition, where you find the LoRa and SigFox MAC and ID.

KanyonKris, 2019-11-07:
Erased the flash with pycom-updater-cli. Now all the NVS commands work.

I saw one post suggesting the NVS errors would be fixed by simply reformatting the the flash partition. I tried this but it didn't work: os.mkfs('/flash')

mc404 - I too was concerned when pycom-updater-cli warned me addresses would be removed, so I wrote them down (from one of the last screens of M5Burner). After erasing, and loading the latest firmware with M5Burner, all of the correct addresses were restored without me having to do anything.

closed 2021-04-16 by KanyonKris