pycom/pycom-micropython-sigfox #355: 1.20.0.rc13 legacy firmware OSError WLAN

drewhepp, 2019-11-02:
I am using a Pycom FiPy (firmware 1.20.0.rc13) and a Pytrack base.

After executing `wlan = WLAN(mode=WLAN.STA)`, `wlan.scan()` results in:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Scan operation Failed!
```
and `wlan.connect('$ssid', auth=(WLAN.WPA2, '$pass'), timeout=5000)` (where $ssid and $pass are my actual network credentials) results in:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: the requested operation failed
```

drewhepp, 2019-11-02:
These errors do not occur with legacy firmware 1.19.0.b5

renzoe, 2020-02-25:
I confirm too.
https://github.com/pycom/pycom-micropython-sigfox/issues/396

seems long issue with 1.2x legacy? I am running a LoPy4

salal-m, 2020-05-08:
Please use the latest 1.20.2 release, I am not able to reproduce the issue on it. @ahepp can you check if it is working for you?

robert-hh, 2020-05-08:
I have no issues with WiFi and the 1.20.2RC6 release. But I'm using my own firmware build in the basic Variant (without PyBytes code).

robert-hh, 2020-05-08:
The documentation here reveals some more parameters to scan:
https://development.pycom.io/firmwareapi/pycom/network/wlan/
E.g. using wlan.scan(scantime=2000) works on devices which timed out before on a rc6 device
on a rc7 device, the parameter set was: `wlan.scan(type=wlan.SCAN_PASSIVE, scantime=5000)`

bervoiv, 2020-06-02:
This is still a problem, any update on a possible fix?
It's rendering my lopy devices useless...

salal-m, 2020-06-02:
Which firmware version are you using?

bervoiv, 2020-06-03:
Hi @salal-m I'm using release='1.20.0.rc13', version='v1.9.4-94bb382 on 2019-08-22' on a LoPy

salal-m, 2020-06-03:
Is there a reason you are not using the latest version? There are no connection issues in 1.20.2.rc7.

salal-m, 2020-06-03:
@bervoiv Please try the latest version. If you don't want to use Pybytes, you can disable it in MicroPython using ** pycom.pybytes_on_boot(False)** (though the additional code for it will still be compiled)
Since we only maintain one version of the firmware now, we can only provide support on that.

bervoiv, 2020-06-03:
@salal-m I thought the legacy was for when we don't need pybytes. And the PyBytes version was for that platform specifically. Thx for your input!

BetterAutomations, 2021-05-07:
FYI I was forced to use 1.20.0.r13 due to [this bug](https://github.com/pycom/pycom-libraries/tree/master/GoogleIOT) on all newer versions I had tested. The [Pycom GoogleIOT plugin](https://github.com/pycom/pycom-micropython-sigfox/issues/545) requires at least 1.20.0.r11 but that also exhibits this bug.

So I am completely and totally stuck on this one. Can't upgrade to newer versions, can't use this version. Surely I'm not the only person?