pycom/pycom-micropython-sigfox #357: WiPy: SOCK_RAW not defined in the WiPy firmware

robert-hh, 2019-11-07:
For an unknown reason, the symbol SOCK_RAW is not defined in the WiPy branch. Using the integer 3 (which is it's value) in the respective situation works. So may I ask you not to exclude WIPy in the definition of SOCK_RAW. The place to go is here: https://github.com/pycom/pycom-micropython-sigfox/blob/Dev/esp32/mods/modusocket.c#L1017

oligauc, 2019-11-12:
At the moment SOCK_RAW should only be using with Lora and Sigfox.

robert-hh, 2019-11-12:
The question is not, whether Pycom software is using it, but whether user software can use it. It is just the symbol definition, that is missing. Supplying 3 as the value of that symbol works.

oligauc, 2019-11-12:
Thanks @robert-hh. We will test and put the constant for the wifi also.

closed 2022-01-26 by robert-hh