pycom/pycom-micropython-sigfox #358: Entering pybytes device token in software

secirugy, 2019-11-09:
Recently, I had to upgrade the FiPy v1.2 with custom (unofficial) firmware build 1.20.1.1r using _pycom_firmware_update_1.16.1.exe_
`Pycom MicroPython 1.20.1.r1 [15b6d69] on 2019-11-02; FiPy with ESP32`
Since I had to select the file on my PC the check box for forcing supplying pybytes token is not showing and therefore the whole step of providing pybytes token was skipped.
So the question is, how can I enter the token through the firmware that I am writing for FiPy? Is there any API for that? How the code should look like? do I need to import pybytes library?

oligauc, 2019-11-12:
Create a file in flash named: pybytes_config.json, with contents

{
  "ota_server": {
    "port": 443,
    "domain": "software.pycom.io"
  },
  "lora": {
    "otaa": {
      "app_key": "OTAA_KEY",
      "app_eui": "OTAA_EUI",
      "app_device_eui": "OTAA_DEVICE_EUI"
    },
    "abp": {
      "app_skey": "ABP_SKEY",
      "nwk_skey": "ABP_NWK_SKEY",
      "dev_addr": "ABP_DEV_ADDR"
    }
  },
  "username": "",
  "device_id": "",
  "server": "mqtt.pybytes.pycom.io",
  "ssl": false,
  "ssl_params": {
    "ca_certs": "/flash/cert/pycom-ca.pem"
  },
  "wifi": {
    "ssid": "",
    "password": ""
  },
  "network_preferences": [
    "wifi"
  ],
  "pybytes_autostart": true,
  "connection_watchdog": true
}

Put the token in the device id section

secirugy, 2019-11-12:
Unfortunately, the latest FiPy firmware 1.20.1.1r was making the device not usable by disabling serial connection to REPL and disabling AP (which was a default WiFi on startup). Therefore, I had to flash another private build by andreas (Pycom.io forum) which had Pybytes excluded from the build...
However, your info can be very useful in the future...

amotl, 2019-11-13:
Dear @secirugy,

> So the question is, how can I enter the token through the firmware that I am writing for FiPy? Is there any API for that? How the code should look like? do I need to import pybytes library?

To extend @oligauc's answer on that with respect to your questions:

Yes, you will have to put in the Pybytes libraries on your own behalf. While the appropriate place should obviously be [1], this looks very much outdated. So, [2] is the right place to obtain them.

I believe it will take some massage to get them added on top but it should certainly be doable. If you combine this with a `pybytes_config.json` as outlined by @oligauc, you should be able to achieve the same functionality.

Cheers,
Andreas.

[1] https://github.com/pycom/pycom-libraries/tree/master/pybytes
[2] https://github.com/pycom/pycom-micropython-sigfox/tree/Dev/esp32/frozen/Pybytes