pycom/pycom-micropython-sigfox #361: Guru Meditation Error on wlan.deinit()

RLars, 2019-11-12:
I am using LoPy4 in the following firmware configuration:
sysname='LoPy4', nodename='LoPy4', release='1.20.1.r1', version='v1.11-3138a13 on 2019-10-08', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.1.3'

Since updating from '1.18.2.r1 the following code loaded as main.py fails. The code mainly disables WiFi. I always end up in an Guru Meditation Error followed by a reset:

```
import time
print("Startup on: {0}".format(os.uname()))
print("================================================================================")

print('Deinitializing WiFi...')
#import pycom
#pycom.wifi_on_boot(False)
from network import WLAN
wlan = WLAN()
wlan.deinit()

while 1:
    time.sleep(5)
```

But when manually giving the following commands in REPL, it works fine:
```
>>>from network import WLAN
>>>wlan = WLAN()
>>>wlan.deinit()
>>>time.sleep(5)
```

Error output:

```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2136
load:0x4009fa00,len:19880
entry 0x400a05d0
Smart Provisioning started in the background
See https://docs.pycom.io/smart for details
Startup on: (sysname='LoPy4', nodename='LoPy4', release='1.20.1.r1', version='v1.11-3138a13 on 2019-10-08', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.1.3')
================================================================================
Deinitializing WiFi...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400911c3  PS      : 0x00060133  A0      : 0x800898a0  A1      : 0x3ffb7140
A2      : 0x00000000  A3      : 0x00060120  A4      : 0x00000001  A5      : 0x0000cdcd
A6      : 0xb33fffff  A7      : 0x3ffaff34  A8      : 0x0000abab  A9      : 0x3ffb7140
A10     : 0x0000cdcd  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x0003f338
A14     : 0x3f800010  A15     : 0x3fbc0bb9  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4009847c  LEND    : 0x400984aa  LCOUNT  : 0xffffffff

Backtrace: 0x400911c3:0x3ffb7140 0x4008989d:0x3ffb7170 0x4008745d:0x3ffb7190 0x4013d922:0x3ffb71d0 0x400d2460:0x3ffb71f0

================= CORE DUMP START =================
<Removed core dump>
================= CORE DUMP END =================
Rebooting...
```

oligauc, 2019-11-12:
@RLars Please can you do an erase all with the Pycom firmware updater and retry ?

RLars, 2019-11-13:
Okay, i did this:
```
>pycom-fwtool-cli.exe --verbose --port COM17 erase_all
Running in PIC mode
Erasing the board can take up to 40 seconds.
Product ID: 18 HW Version: 2 FW Version: 0.0.8
Connecting....
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Auto-detected Flash size: 8MB
Board erased successfully in 11.09 seconds
```

Then updated again with pycom-fwtool.exe:
[image]
[image]

(Skipped Sigfox registration)

Result:
[image]

Unfortunately i don't know, where to download the current firmware to manually upload it via CLI utility. In https://software.pycom.io/downloads/LoPy4.html it seems not to be contained.

After the update process and uploading the main.py, the problem is still there:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2136
load:0x4009fa00,len:19880
entry 0x400a05d0
Smart Provisioning started in the background
See https://docs.pycom.io/smart for details
Startup on: (sysname='LoPy4', nodename='LoPy4', release='1.20.1.r1', version='v1.11-3138a13 on 2019-10-08', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.1.3')
================================================================================
Deinitializing WiFi...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400911c3  PS      : 0x00060133  A0      : 0x800898a0  A1      : 0x3ffb7140
A2      : 0x00000000  A3      : 0x00060120  A4      : 0x00000001  A5      : 0x0000cdcd
A6      : 0xb33fffff  A7      : 0x3ffaff34  A8      : 0x0000abab  A9      : 0x3ffb7140
A10     : 0x0000cdcd  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x0003ea44
A14     : 0x3fbc0020  A15     : 0x3fbc0f4d  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4009847c  LEND    : 0x400984aa  LCOUNT  : 0xffffffff

Backtrace: 0x400911c3:0x3ffb7140 0x4008989d:0x3ffb7170 0x4008745d:0x3ffb7190 0x4013d922:0x3ffb71d0 0x400d2460:0x3ffb71f0

================= CORE DUMP START =================
<removed core dump>
================= CORE DUMP END =================
Rebooting...
```

robert-hh, 2019-11-13:
It seems that the trouble is caused by pybytes activities. Many people report errors. Until eventually this bug has been fixed by Pycom, you can disable autostart of pybyes by adding a configuration file to you system. Unfortunately, it has to boot, which it does not at the moment. The configuration file is called pybytes_config.json, and the content should be:
```
{"pybytes_autostart": false}
```
To get that to the device, you have to install a working firmware first, e.g. 1.20.0.rc13 from the download page. Then you can upload this file and re-install 1.20.1.r1. You may also use one of the packages @andreas has collected which do not use pybytes. Look for instance at this post: https://forum.pycom.io/topic/5360/fipy-v1-2-problems/12

amotl, 2019-11-13:
Dear @RLars,

if you feel lucky, you might want to try our custom build just released [1]. More background about this is available through [2].

If we are both lucky, this will improve the stability significantly. If you will be still receiving the core dumps, I will be happy if you would share its content with me. It is the section you've removed and marked with `<Removed core dump>`.

Please be aware that you will have to erase your device completely before flashing in order to keep things straight. You will find respective references to this on the forum. Hint: Use `pycom-fwtool-cli --port /dev/ttyUSB0 erase_all`, see also [3].

With kind regards,
Andreas.

P.S.: Thanks @robert-hh for the introduction.

[1] https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/LoPy4-1.20.1.r1-0.6.0-vanilla-dragonfly.tar.gz
[2] https://community.hiveeyes.org/t/investigating-random-core-panics-on-pycom-esp32-devices/2480
[3] https://community.hiveeyes.org/t/installing-the-recent-pycom-firmware-1-20-1-r1-requires-erasing-the-flash-memory-completely/2688

RLars, 2019-11-13:
Hi Andreas,

> if you feel lucky, you might want to try our custom build just released [1]. More background about this is available through [2].
> 
> If we are both lucky, this will improve the stability significantly.

yes, it works:-) :
```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2136
load:0x4009fa00,len:19860
entry 0x400a05d0
Startup on: (sysname='LoPy4', nodename='LoPy4', release='1.20.1.r1-0.6.0-vanilla-dragonfly', version='69dd8b5d-dirty on2019-11-13', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')
================================================================================
Deinitializing WiFi...
```

Should i close this issue or do you want to use it to fix the bug in the main release?

RLars, 2019-11-13:
Now i have a different problem:
```
>>>from network import LoRa
>>>lora = LoRa(mode=LoRa.LORAWAN, region=LoRa.EU868, tx_power=14, >>>power_mode=LoRa.ALWAYS_ON)
>>> lora.mac()
b'\xff\xff\xff\xff\xff\xff\xff\xff'
```

The LoRa MAC cannot be read out anymore.

Isn't this supported by 1.20.1.r1-0.6.0-vanilla-dragonfly?

amotl, 2019-11-13:
Dear @RLars,

> yes, it works:-)

Thanks for letting us know. Feel free to close this issue. I will try to work on getting this mainlined ASAP.

We are kindly asking to report back to us if you see any other glitches with this specific build. In the meanwhile: Enjoy your further development.

With kind regards,
Andreas.

cc @robert-hh, @Xykon, @danicampora.

amotl, 2019-11-13:
> Now i have a different problem: The LoRa MAC cannot be read out anymore.

While we didn't break that on purpose within this unofficial build, unfortunately I can't tell anything about this right now as I don't own a LoPy4 device. I believe it would be good to check against a recent official build and then open a different issue about this topic.

Sorry that I can't help on this specific thing!

robert-hh, 2019-11-13:
> Now i have a different problem:

That data was erased by the erase all procedure. To restore it, you can go two ways:
a) If you still know the value the LoraMac had before (you knew it for your post above), then you can follow the procedure explained here:https://forum.pycom.io/topic/1272/lora-mac-ffffffffffffffff/21
b) If you do not know it, you have to use the pycom updater, which will restore it. 
b.1)Use the GUI version pycom updater to install a recent image (like pybytes 1.20.1) from the internet. You do not have to use it, Just the download process restores the Lora MAC and SIgFox ID.
b.2) Use the pycom updater to flash the vanilla non-pybytes image. Do not erase the flashin between.

amotl, 2019-11-13:
@robert-hh: Thanks for outlining this procedure to @RLars. As I don't own a LoPy4 device, I haven't been aware of this important detail. Apologies!

RLars, 2019-11-13:
Thanks, i just performed an "erased_all" one more time and flashed the hiveeyes firmware with the gui instead of the CLI. Now it works!

Thanks all

closed 2019-11-13 by RLars

deltazita, 2020-02-24:
I used to have the same issue with v1.20.1 and I solved it by writing :
wlan = WLAN(mode=WLAN.STA)
wlan.deinit() 

instead of 

wlan = WLAN()
wlan.deinit()
I hope it works for you as well