pycom/pycom-micropython-sigfox #370: mpy-cross build error with macos 10.14.6 and GNU make

tcados, 2019-11-27:
I am getting an error messages when building mpy-cross with Mac OS 10.14.6 and GNU make (I've tried both 3.81 and 4.2)

`
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Traceback (most recent call last):
  File "../py/makeversionhdr.py", line 91, in <module>
    make_version_header(sys.argv[1])
  File "../py/makeversionhdr.py", line 66, in make_version_header
    git_tag, git_hash = info
TypeError: 'NoneType' object is not iterable
make: *** [../py/py.mk:328: build/genhdr/mpversion.h] Error 1
`

I've been successful in building mpy-cross in other MicroPython repositories, so I am led to believe there is an issue here.

msariisik, 2020-05-11:
Can you give us the device information and which release of pycom-micropython-sigfox are you using?

rouzbeh-afrasiabi, 2021-02-16:
Same issue, trying to build on Windows subsystem through bash. pycom-micropython-sigfox version is 1.20.2.r4.

William04A, 2022-06-01:
Hi! Today, I experienced an error **similar** to this too. I spent about an hour tracking it down, so I thought I'd write a summary in case it helps. The error I got was identical to the one mentioned [here](https://github.com/adafruit/circuitpython/issues/791).

The root problem (for me at least) is that the build scripts are trying to get information about the git repository in the `pycom-micropython-sigfox` directory while trying to compile `mpy-cross`, but for some reason, this fails and the fallback because of that is to look in a file called "conf.py" as specified in the building files for mpy-cross. However, since this file doesn't exist (looks like it has been removed from the repository), the fallback doesn't work, causing the build to fail.
(you can find the code that is throwing this error [in this file](https://github.com/pycom/pycom-micropython-sigfox/blob/Dev/py/makeversionhdr.py) (see the `get_version_info_from_git()` function (at the time of this writing on line 14 in the file) and also see the definition for the `make_version_header` function (at the time of this writing on line 60 in the same file)).

You don't get much error information, so your best bet is to add some debugging yourself which can be done by editing the
pycom-micropython-sigfox/py/makeversionhdr.py file. It has been linked above. For example, what I did was adding `print`-functions to all places in `get_version_info_from_git()`  where the code handled the errors. Then, I tried `sudo make` again until I got output. 

For me, the problem was eventually tracked down to the cloned GitHub repository not being owned by `root`, which caused errors when using `sudo make` for `mpy-cross`. The final error might be different for you, but hopefully, I helped someone to at least debug the issue.