pycom/pycom-micropython-sigfox #378: Two core dumps

tschudin, 2019-12-31:
```
Pycom MicroPython 1.20.1.r2 [v1.11-06dfad0] on 2019-11-30; LoPy4 with ESP32
Pybytes Version: 1.3.0
```

After some random time (one case was after approx 2 hours), I got two different core dumps. My program uses LoRa (raw), WLAN in STA mode, and a timer alarm to send a packet every two seconds.

```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0xfa11a504  PS      : 0x00060630  A0      : 0xfa11a504  A1      : 0x3ffd3820
A2      : 0x3ffcc5e4  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffe3e18  A7      : 0x00000080  A8      : 0xb33fffff  A9      : 0x0000cdcd
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060020  A13     : 0x3ffd38c8
A14     : 0x00000017  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000014
EXCVADDR: 0xfa11a504  LBEG    : 0x4009847c  LEND    : 0x400984aa  LCOUNT  : 0xffffffff

Backtrace: 0x7a11a504:0x3ffd3820 0x7a11a501:0x3ffd3850 0x4014b109:0x3ffd3880 0x4020865b:0x3ffd38b0

================= CORE DUMP START =================
8DcAABEAAABsAQAA
...
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
================= CORE DUMP END =================
Rebooting...
```

and

```
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014ba58: d32142ac ffdb8173 980020c0
Core 0 register dump:
PC      : 0x4014ba5f  PS      : 0x00050830  A0      : 0x8014b10c  A1      : 0x3ffd3bbc
A2      : 0x3ffe3cc0  A3      : 0x00000000  A4      : 0x4014a71a  A5      : 0x00000000
A6      : 0x3ffcc50d  A7      : 0x00060620  A8      : 0x8014adcd  A9      : 0x00000000
A10     : 0x3ffcc5e4  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000080  SAR     : 0x00000020  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4009847c  LEND    : 0x400984aa  LCOUNT  : 0xffffffff

Backtrace: 0x4014ba5f:0x3ffd3bbc

================= CORE DUMP START =================
hDQAABAAAABsAQAA
...
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
================= CORE DUMP END =================
E (16428) esp_core_dump: Skipped 1 tasks with bad TCB!
E (16433) esp_core_dump: Crashed task has been skipped!
Rebooting...
```

amotl, 2020-01-01:
Dear @tschudin,

can you try to use the "dragonfly" builds as outlined on [1] or #361 and report back if you see any improvements?

However, you might observe the same thing as @pascalschaefer outlined within [2]. In this case, these builds won't help either. I will be happy to review appropriate core dumps.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/testing-the-custom-dragonfly-builds-on-pycom-devices/2746
[2] https://community.hiveeyes.org/t/core-panics-with-the-dragonfly-build/2878

tschudin, 2020-01-03:
Hi Andreas (@amotl)

The dragonfly image did not improve things. On the other hand you now have two core dumps that you were looking for..

best, c

```
(rx_timestamp=1987271930, rssi=-69, snr=6.0, sfrx=7, sftx=7, tx_trials=0, tx_power=20, tx_time_on_air=31, tx_counter=5287, tx_frequency=915000000)
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0xd981b901  PS      : 0x00060230  A0      : 0x8020b076  A1      : 0x3ffd3880
A2      : 0x3ffe3c9c  A3      : 0x00000001  A4      : 0x00000574  A5      : 0x3ffd3968
A6      : 0x00000017  A7      : 0x00000000  A8      : 0x8020aa58  A9      : 0x00000000
A10     : 0x3ffcc838  A11     : 0x3ffe3cd8  A12     : 0xb33fffff  A13     : 0x0000cdcd
A14     : 0x00000001  A15     : 0x3ffbe680  SAR     : 0x00000020  EXCCAUSE: 0x00000014
EXCVADDR: 0xd981b900  LBEG    : 0x40098868  LEND    : 0x40098884  LCOUNT  : 0xffffffff

Backtrace: 0x5981b901:0x3ffd3880 0x4020b073:0x3ffd38b0 0x4014bf19:0x3ffd38f0 0x4014c06e:0x3ffd3920 0x4020c85b:0x3ffd3950

================= CORE DUMP START =================
UDsAABEAAABsAQAA
BDr9P8A3/T/8Of0/
MDj9P5A5/T8SwiZsvEL8P7xC/D8EOv0/tEL8PwIAAADA9/0/wPf9PwQ6/T8AAAAA
FwAAAAAs/T93aWZpAPFXOaPUwELC8fwAAAAAAPw5/T8AAAAAIAIGABcAAAABAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
n5JAPwAAAACokQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPaMFA==
TO0DAAG5gdkwAgYAdrAggIA4/T+cPP4/AQAAAHQFAABoOf0/FwAAAAAAAABYqiCA
AAAAADjI/D/YPP4///8/s83NAAABAAAAgOb7PyAAAAAUAAAAALmB2WiICUCEiAlA
////////P7PNzQAAIDkIQFhK+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAANA4/T8AAAAA/////2j1+z8cvxSAsDj9P9g8/j8AAAAA
//8/swAAAABACxVAXgsVQAMAAACEOwhA3EL8PwQtCUBxwBSA8Dj9PwAAAAAFAAAA
//8/swAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAhMb8PwAAAAAAAQAA
XsgggCA5/T8AAAAAAAAAAAAAAACReqEAbBP8P3gU/D8AAEAAAQAAAP4AAAABAAAA
AAAAAFA5/T9sE/w/eBT8PwAAAABQOf0/bBP8P3gU/D9Y//w//////wEAAABY//w/
AAAAAJA5/T8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAXAAAAAAAAAAAAAAAAAAAA
AAAAAFgD/T8aAAAAcBP8PwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAJw5/T8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
ML77P5C8+z8ovvs/
kLz7P8C9+z+I3zKQ8ED8P8S4+z8wvvs/6ED8PxkAAAAsbLuqkHA9MjC++z8AAAAA
AAAAACy6+z9JRExFMQCsAewt8yMMC+cAAQAAACi++z8AAAAAIwAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
n5JAPwAAAACokQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIHpWg==
hDkIQLJtIEAwDwYAXsUWgFC9+z8AAAAA8ED8P0AAAACA8Ps/yLH8P4yx/D9A5f0/
jOX9PyEABgAAAAAAoYOfAGAT/D+ZYJ8AAAAAAAAAAABQvfs/AAAAAAAAAAAAAAAA
AAAAAIQ7CEBgE/w/BC0JQIjO+D8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAP//P7MAAAAAAAAAAAAAAAAMGwmAcL37PyAAAAABAAAA
//8/swAAAAAAAAAAAAAAAAAAAACQvfs/AAAAAHhA/D8gAAAASNv8P+BC/D8AAAAA
AAAAAMC9+z8AAAAAAAAAAGj1+z8AAAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAzL37PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
vLj7PyC3+z+0uPs/
ILf7P1C4+z/wJZ7MOL77P/BA/D+8uPs/6ED8PxkAAAAXsgJhCgKIILy4+z8AAAAA
AAAAALi0+z9JRExFMABVUSdioUHt254AAAAAALS4+z8AAAAAIwAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
n5JAPwAAAACokQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAX2Dw==
hDkIQLJtIEAwAgYAXsUWgOC3+z8AAAAAUA8AAAAAAACgGf0/AwAAACMABgDNBAAA
[... 651 lines cut ...]

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2136
load:0x4009fa00,len:19860
entry 0x400a05d0
Starting boot.py ...
  before LoRa
  before callback
  before sender
boot.py ends
Starting main.py ...
Device:       192.168.1.11
LoRa MAC:     70:b3:d5:49:9b:2c:83:f3
LoRa freq:    915000000
LoRa bw:      125kHz
LoRa coding:  4_6
LoRa stats:   (rx_timestamp=0, rssi=0, snr=0.0, sfrx=0, sftx=0, tx_trials=0, tx_power=20, tx_time_on_air=0, tx_counter=0, tx_frequency=0)
Pycom MicroPython 1.20.1.r1-0.7.0-vanilla-dragonfly-onewire-i2s [daf40f36-dirty] on 2019-12-04; LoPy4 with ESP32
Type "help()" for more information.
>>>
```

amotl, 2020-01-04:
### Coredump 1
```
==================== CURRENT THREAD STACK =====================
#-1 0x5981b901 in ?? ()
Backtrace stopped: Cannot access memory at address 0x5981b901

======================== THREADS INFO =========================
  Id   Target Id         Frame
  17   process 16        0x4000bff0 in ?? ()
  16   process 15        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073495820, uxInitialCount=0, pxStaticQueue=0x3ffc3f58 <vflash_mem+1900>) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  15   process 14        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073460044, uxInitialCount=1073603568, pxStaticQueue=0x3ffbb398) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  14   process 13        0x4000bff0 in ?? ()
  13   process 12        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073444348, uxInitialCount=0, pxStaticQueue=0x3ffb7648) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  12   process 11        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073458244, uxInitialCount=1073580336, pxStaticQueue=0x3ffbac90) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  11   process 10        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073445832, uxInitialCount=0, pxStaticQueue=0x3ffb7c14) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  10   process 9         0x4000bff0 in ?? ()
  9    process 8         0x40092b55 in xTaskNotify (xTaskToNotify=0x0, ulValue=<optimized out>, eAction=<optimized out>) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4872
  8    process 7         0x4009192e in vTaskExitCritical (mux=0xa17a92) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4274
  7    process 6         0x4009192e in vTaskExitCritical (mux=0xa17cb8) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4274
  6    process 5         xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073468924, uxInitialCount=1073458080, pxStaticQueue=0x3ffbd648) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  5    process 4         0x4009192e in vTaskExitCritical (mux=0xa17a92) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4274
  4    process 3         0x4009192e in vTaskExitCritical (mux=0xa17a92) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4274
  3    process 2         0x40206db2 in _strtoull_r (rptr=0x8000, nptr=0xf50 <error: Cannot access memory at address 0xf50>, endptr=<optimized out>, base=1073551776) at ../../../.././newlib/libc/stdlib/strtoull_r.c:93
  2    process 1         0x40206db2 in _strtoull_r (rptr=0x0, nptr=0x3ffc40f0 <s_timer_semaphore_memory+80> \"\", endptr=<optimized out>, base=1073475712) at ../../../.././newlib/libc/stdlib/strtoull_r.c:93
* 1    <main task>       0x5981b901 in ?? ()
```

### Coredump 2
```
==================== CURRENT THREAD STACK =====================
#0  xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073603904, uxInitialCount=0, pxStaticQueue=0x3ffde58c) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

======================== THREADS INFO =========================
  Id   Target Id         Frame
  16   process 15        0x4000bff0 in ?? ()
  15   process 14        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073460044, uxInitialCount=1073603568, pxStaticQueue=0x3ffbb398) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  14   process 13        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073495820, uxInitialCount=0, pxStaticQueue=0x3ffc3f58 <vflash_mem+1900>) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  13   process 12        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073608596, uxInitialCount=1073559904, pxStaticQueue=0x3ffdf7e0) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  12   process 11        0x4000bff0 in ?? ()
  11   process 10        xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073444348, uxInitialCount=0, pxStaticQueue=0x3ffb7648) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  10   process 9         xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073458244, uxInitialCount=1073580336, pxStaticQueue=0x3ffbac90) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  9    process 8         xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073445832, uxInitialCount=0, pxStaticQueue=0x3ffb7c14) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  8    process 7         0x4000bff0 in ?? ()
  7    process 6         0x40092b55 in xTaskNotify (xTaskToNotify=0x0, ulValue=<optimized out>, eAction=<optimized out>) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4872
  6    process 5         0x4009192e in vTaskExitCritical (mux=0xe7046) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4274
  5    process 4         xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073468924, uxInitialCount=1073458080, pxStaticQueue=0x3ffbd648) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
  4    process 3         0x4009192e in vTaskExitCritical (mux=0xe7047) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4274
  3    process 2         0x4008169e in esp_crosscore_int_send_yield (core_id=1) at /Users/iwahdan/esp/pycom-esp-idf/components/esp32/crosscore_int.c:112
  2    process 1         0x4009192e in vTaskExitCritical (mux=0xe7046) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/tasks.c:4274
* 1    <main task>       xQueueCreateCountingSemaphoreStatic (uxMaxCount=1073603904, uxInitialCount=0, pxStaticQueue=0x3ffde58c) at /Users/iwahdan/esp/pycom-esp-idf/components/freertos/queue.c:659
```

amotl, 2020-01-04:
Looking for `esp_crosscore_int_send_yield` on the Espressif issue tracker [1], some of the things listed there might resonate with you.

However, all things are from 2017 and should be fixed already.

[1] https://github.com/espressif/esp-idf/search?q=esp_crosscore_int_send_yield

amotl, 2020-01-15:
Dear @tschudin,

while I don't have any solution for this yet, I believe you, @pascalschaefer and @ifiwaspresident might be observing the same issue on their LoPy4 devices. See also https://github.com/pycom/pycom-micropython-sigfox/issues/381 and https://community.hiveeyes.org/t/investigating-core-panics-on-the-lopy4/2878.

May I humbly ask if you are also using alarm timers, radio interrupts and writing to files / writing to NVRAM within your firmware?

With kind regards,
Andreas.

tschudin, 2020-01-15:
Hi @amotl 

Things have improved a lot since I do not use (event) callbacks anymore - I never did any NVRAM actions or radio triggers. Now I run my own central loop and use poll() to get notified when I can read a new packet, or send the next packet (the socket is always in non-blocking mode). Core dumps have become very rare since then.

(but other problems now become visible: file system-related and LoRa getting numb after a few minutes ...)

amotl, 2020-01-15:
Dear @tschudin,

> Things have improved a lot since ...

Thanks for letting me know.

> But other problems now become visible: File system-related and LoRa getting numb after a few minutes.

Interesting. Please also let me know how this turns out and feel free to ask. We have also been putting in some hours of debugging into this alrady.

> File system-related

May I ask whether you are using LittleFS already or if you are still on FatFS?

With kind regards,
Andreas.

tschudin, 2020-01-15:
@amotl wrote:

> May I ask whether you are using LittleFS already or if you are still on FatFS?

After three blatant cases where the first 4K of files on FatFS were overwritten with arbitrary content, I have now started to move to LittleFS. I will report back when all nodes are reflashed and I have some longer run time experience.

Re LoRa, I now call init() on the LoRa instance every 20 sec but also have to see whether this improves things.

best, c

amotl, 2020-01-15:
> I have now started to move to LittleFS.

Good choice!

> I will report back when all nodes are reflashed and I have some longer run time experience.

Thanks and good luck!

amotl, 2020-02-12:
Dear @tschudin,

I just wanted to let you know @pascalschaefer had success upgrading to the most recent 1.20.2.r3 firmware release [1,2].

> I tried the new build [LoPy4-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz ](https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/LoPy4-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz) and not any “bad00bad bad00bad bad00bad” Core Panic happend till now.

Thanks @husigeza and the whole Pycom team for mitigating this issue.

With kind regards,
Andreas.

[1] https://forum.pycom.io/topic/5514/random-core-panic-with-lora-socket-send-recv-device-doesnt-restart-try-with-config_esp32_panic_silent_reboot
[2] https://community.hiveeyes.org/t/investigating-core-panics-on-the-lopy4/2878/29

amotl, 2020-02-12:
> but other problems now become visible: file system-related and LoRa getting numb after a few minutes ...

Regarding the filesystem issues, you might want to try the _squirrel_ build referenced above. There are some patches on top of the official Pycom code base which might be able to solve this problem for you. Please recognize the [installation notes for LoPy, LoPy4, SiPy and FiPy devices](https://community.hiveeyes.org/t/dragonfly-firmware-for-pycom-esp32/2746#notes-for-lopy-lopy-sipy-and-fipy-devices). We will be happy about any reports regarding this topic.