pycom/pycom-micropython-sigfox #379: sqns firmware 43818

krismc, 2020-01-07:
My latest GPY arrived with 43818 firmware in the lte modem. Since this new GPY is more stable on lte than my other GPYs I would like to upgrade the 41065 modem firmware in them to 43818. Where can I get a copy of the 43818 sqns firmware please?

catalinio, 2020-01-08:
Hi @krismc,

We're in the process of testing the upgrade packages.

krismc, 2020-01-08:
tnx, any idea when an upgdiff_41065-to-43818.dup file might be available?

abatardi, 2020-06-01:
5 months later no update?

krismc, 2020-06-06:
Yeah it's a pain. I tried some NB-IoT41019 firmware in my latest gpy & now I'm stuck with it, can't roll it back to the CATM1-43818 that came with it.

JCBuck, 2020-06-29:
Been another month? Also wondering where this might be...