pycom/pycom-micropython-sigfox #380: LoPy4,WiPy3 loosing firmware on power failure

HenroRitchie, 2020-01-11:
Pycom modules used: LoPy4 and WiPy3
Firmware Version: Latest Legacy (I think it is legacy v1.20.0.rc13)

I don't have the device with me so I cannot check.

* Exact steps to cause this issue
    1. It is difficult to exactly name the steps, but removing the power at random intervals sometimes causes the Pycom module to lose the custom firmware and restores the default firmware.
* What you expected
    * Pycom modules should keep the custom loaded firmware under ALL circumstances.
* What happened instead
    * Pycom module replaced custom firmware with default firmware.

As mentioned - I don't have an exact method to reproduce the problem but have experienced this numerous times. I programmed a Pycom module yesterday and shipped it to a client - when he turned it on the custom firmware was replaced with the default vanilla firmware as if the device have just been formatted.

Thank you!

amotl, 2020-01-15:
Dear @HenroRitchie,

you should definitively be using LittleFS these days. FatFS is well known to regularly corrupt the filesystem so the device bounces it.

With kind regards,
Andreas.

amotl, 2020-02-19:
Dear Henro,

may I humbly ask if you are accessing the flash memory through the filesystem at runtime? If so, these bits of additional information might help you along.

Several people have been able to mitigate similar issues to what @mfallavol is observing within #402 with one of our custom/unofficial Dragonfly builds based on Pycom’s 1.20.1.r1 [1]. We've refreshed this the other day and published corresponding Squirrel builds based on Pycom’s most recent 1.20.2.rc3 [2].

If you are lucky to try one of these images, we will be happy to hear about the outcome for you.
- [WiPy-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz](https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/WiPy-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz)
- [LoPy4-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz](https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/LoPy4-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz)

Please read both the [installation instructions](https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960#installation-instructions) and the [notes for LoPy, LoPy4, SiPy and FiPy devices](https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960#notes-for-lopy-lopy-sipy-and-fipy-devices) carefully, a) regarding the `erase_all` procedure and b) for restoring the Lora MAC after the `erase_all` procedure.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/dragonfly-firmware-for-pycom-esp32/2746
[2] https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960

HenroRitchie, 2020-02-19:
Dear Andreas

Thank you for your reply. Since your previous post, I've moved to LittleFS. I've not pushed any new units into production as we are waiting for the boards from the factory. When we receive them, which should be by the end of March, I will have 15+ units to work with.

The unit in the field, that was troublesome previously, have been working just fine.

I will then use your advice to load the firmware and test.

Best Regards
Henro Ritchie