pycom/pycom-micropython-sigfox #381: LoPy 4 Guru Meditation Error

ifiwaspresident, 2020-01-14:
Hey there,

I have a LoPy 4 with an extension board. 
I am reading a sensor and send the measurement via LoRaWAN. 
It runs good for maybe 10 minutes or so when the loop reaches this line it crashes:

            lora.nvram_save()

I wrapped that line in a try statement but it does not catch it. Is there something I can do or is it some low level problem?

**Board description:**
Pycom MicroPython 1.20.1.r2 [v1.11-06dfad0] on 2019-11-30; LoPy4 with ESP32
Pybytes Version: 1.3.0
Type "help()" for more information.
>>>
>>>
Uploading project (main folder)...
Safe booting device... (see settings for more info)

Uploading to /flash...
Reading file status
[1/1] Writing file main.py (3kb)
Upload done, resetting board...
OKets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2136
load:0x4009fa00,len:19880
entry 0x400a05d0
Smart Provisioning started in the background
See https://docs.pycom.io/smart for details

**Error Code:**
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40202728: bad00bad bad00bad bad00bad

robert-hh, 2020-01-14:
In order to isolate effects, you could try to disable pybytes. For that, you have to create a file on the lopy4 with the name: `pybytes_config.json` and the content
```
{ "pybytes_autostart": false }
```
You may also try to use the firmware build provided by  @andreas. Info e.g. here: https://forum.pycom.io/topic/5560/guru-meditation-error/11

ifiwaspresident, 2020-01-15:
Thanks for the fast reply!

Disabling Pybytes seem to help. The board is working fine for now. 
So this is a nvram access problem? Where Pybytes is writing something to the nvram and lora.nvram_save() does not wait for the other process to finish? (or the other way around)

amotl, 2020-01-15:
> Disabling Pybytes seem to help. The board is working fine for now.

Good to hear! Thanks for pointing out to disable Pybytes, @robert-hh!

> So this is a nvram access problem? Where Pybytes is writing something to the nvram and lora.nvram_save() does not wait for the other process to finish? (or the other way around)

Maybe. May I humbly ask whether you are using LoRa callbacks, threads and/or alarm timers within your MicroPython program?