pycom/pycom-micropython-sigfox #383: Pycom-fwtool fo EOAN

hdhk, 2020-01-16:
Hi,
 
* The Pycom board lopy4 (expansion board 3 (v3.1))
* The firmware version the REPL:

(sysname='LoPy4', nodename='LoPy4', release='1.20.0.r3', version='V1.9.4-398a290 on 2019-09-19', machine='LoPy4 with ESP32', lorawan='1.0.2',)

    1. First I did: dfu-util -D expansion31_0.0.11.dfu 
Result: wrong mode ef99 , not ef98 
    2. Searched for different ways to switch to update mode, couldn't find. Looked for tutorials, gave up.
    3. Next pycom-fwtool is NOT for eoan (ubuntu 19.10) available.

Is the first step necessary? If so, how do I get the board into UpdateMode 0x99? With or with out the module on the extension board? When will the version of the pycom-fwtool for Ubuntu 19.10 be released? 

Best regards, HDHK

catalinio, 2020-01-16:
hi @hdhk 
The dfu update is not required, most likely you have already the last Expansion Board firmware, released 1 year ago. 
Fyi, `wrong mode ef99 , not ef98` is because the board is not in update mode, probably the steps explained here https://docs.pycom.io/pytrackpysense/installation/firmware/#using-dfu-util-with-pytrack-pysense-and-expansion-board-v3 are not done (and update mode is just for 7-8 seconds).

We'll look into releasing Firmware Update tool for Ubuntu 19.10.
Could you try the generic linux release from https://software.pycom.io/downloads/linux-1.16.1.html?

Thanks,
Catalin

hdhk, 2020-01-16:
Thanx Catalino, works, HDHK

hdhk, 2020-01-18:
Hi Catalin, I have trouble with that linux release in case of fonts. I get a "white" screen. BR HDHK

catalinio, 2020-01-20:
Which glib version did you tried? Could you try them both?
Anyway, we'll try releasing the Ubuntu 19 version.

thanks

hdhk, 2020-01-20:
GLIBC 2.18 amd64, trouble with fonts, choose , back to (installed 18.04 LTS) ubuntu 
Works and fine. Now I have 2 LoPY4 with pymesh running.

Next Step: Creation of Border Router as wifi AP.
Do you have some hints? Unfortunally I‘m Not an developer, sorry.
B hdhk