pycom/pycom-micropython-sigfox #385: [lopy4] get WiFi ssid

nilay994, 2020-01-20:
(sysname='LoPy4', nodename='LoPy4', release='1.20.1.r2', version='v1.11-06dfad0 on 2019-11-30', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.3.0')

Error while running `pymakr > Get WiFi SSID` on vscode
```
>>> wlan.mac()
(sta_mac=b'\xb4\xe6-\xf2;\xe1', ap_mac=b'\xb4\xe6-\xf2;\xe2')
>>> hexlify(wlan.mac())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object with buffer protocol required
```

robert-hh, 2020-01-20:
This is no issue. As you can see, wlan.mac() returns a tuple of two mac addresses. So you have to write:
`hexlify(wlan.mac()[0]) `
or 
`hexlify(wlan.mac()[1])`

nilay994, 2020-01-20:
In that case the command sent via pymakr for vscode when executing `Get WiFi SSID` should be 
`from network import WLAN; from binascii import hexlify; from os import uname; wlan = WLAN(); mac = hexlify(wlan.mac()[0]).decode('ascii'); device = uname().sysname;print('WiFi AP SSID: %(device)s-wlan-%(mac)s' % {'device': device, 'mac': mac[len(mac)-4:len(mac)]})`

or 
`from network import WLAN; from binascii import hexlify; from os import uname; wlan = WLAN(); mac = hexlify(wlan.mac()[1]).decode('ascii'); device = uname().sysname;print('WiFi AP SSID: %(device)s-wlan-%(mac)s' % {'device': device, 'mac': mac[len(mac)-4:len(mac)]})`