pycom/pycom-micropython-sigfox #386: FiPy has LoPy firmware when updated with Pycom Firmware Update Tool

jvigliottaa, 2020-01-23:
I used the Pycom Firmware Update Tool and upon install of software my FiPy device now has LoPy software on it so I can not use the LTE module or SIM card on my device. The board shows:

`(sysname='LoPy', nodename='LoPy', release='1.20.1.r2', version='v1.11-06dfad0 on 2019-11-30', machine='LoPy with ESP32', lorawan='1.0.2', pybytes='1.3.0')`

`Pycom MicroPython 1.20.1.r2 [v1.11-06dfad0] on 2019-11-30; LoPy with ESP32`

How can I manually select/load FiPy firmware on to my FiPy device?

Thank you!

robert-hh, 2020-01-23:
Hello @ @jvigliottaa, this is mor a question for the forum, but anyhow: you could try to run the firmware update tool again and select the proper device. If that is not possible, you can download the firmware package as a file and use the option: "Install from a file". The Firmware packages are here: https://software.pycom.io/downloads/FiPy.html
It may help to tick the box: "full erase" in the advances settings. That will lead to a fresh clean device, but will also delete all existing files.
For some reason unknown to me Pycom does not provide the latest revision here. But once you have the proper firmware loaded, you may try to use again the online method for the proper FiPy firmware.

jvigliottaa, 2020-01-23:
@robert-hh I downloaded the latest firmware for the FiPy and put it on the device and it seems to be working but the Pycom Firmware Update Tool still thinks it is a LoPy device.

robert-hh, 2020-01-23:
Then there seems to be a Bug in the Pycom device database. Send a message to support@pycom.io. There might be a response.