pycom/pycom-micropython-sigfox #395: could not find module '_main_pybytes.py'

dane-if3, 2020-02-05:
1.20.2.rc2 [v1.11-605f40f] on 2020-02-04; WiPy with ESP32

upgraded to 1.20.2.rc2 from rc1.
on powerup I now get:
could not find module '_main_pybytes.py'

previously I had a message about pybytes_autostart being disabled
I have { "pybytes_autostart": false } in pybytes_config.py

doniks, 2020-02-05:
Thanks for flagging! We made a new release rc3 that should fix it. Please retest

dane-if3, 2020-02-05:
tested with rc3.

works as expected.
  Pybytes connected successfully (using the built-in pybytes library)
  Pybytes configuration read from /flash/pybytes_config.json
and with  { "pybytes_autostart": false } 
  Pybytes configuration read from /flash/pybytes_config.json
  Not starting Pybytes as auto-start is disabled

closed 2020-02-05 by dane-if3