pycom/pycom-micropython-sigfox #396: legacy firmware causes OSError: Scan operation Failed!, pybytes firmware works.

dane-if3, 2020-02-05:
1.20.0.rc13 [v1.9.4-94bb382] on 2019-08-22; WiPy with ESP32

Running the 'Multiple Networks using a Static IP Address' code from the documentation
(https://docs.pycom.io/tutorials/all/wlan/#app )
results in the following:
```
Starting WiFi
Scanning for known wifi nets
Traceback (most recent call last):
  File "boot.py", line 30, in <module>
OSError: Scan operation Failed!

```
The same code on 1.20.2rc3 runs as expected
This problem existed on earlier versions of 1.20.0.rc1x [legacy] and not on 1.20.x [pybytes]

amotl, 2020-02-12:
Dear @dane-if3,

if you need this, [WiPy-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz](https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/WiPy-1.20.2.rc3-0.8.0-vanilla-squirrel.tar.gz) would be the corresponding non-Pybytes build until Pycom releases corresponding firmware images.

With kind regards,
Andreas.

dane-if3, 2020-02-13:
Thanks, I can confirm it works in this version.
Regards, 
Dane

JDWarner, 2020-04-25:
Also having this issue, frustrating.  Please update the official non-Pybytes firmware.