pycom/pycom-micropython-sigfox #400: LoPy4 REPL not working in 1.20.2.rc3

tschudin, 2020-02-15:
LoPy4 with expansion board v3.1

I'm stuck with 1.20.1 because the release candidates do not have the REPL working (1.20.1.rc* as well as 1.20.2.rc*), I tried with two different serial programs. When going back to 1.20.1, things work again.
Specifics: When upgrading, I'm selecting LittleFS and erasing the flash. Running on MacOS.
Would love to try out 1.20.2.rc3 but can't because of this.

husigeza, 2020-02-18:
Hello @tschudin ,
Did you also try using pymakr ?

tschudin, 2020-02-18:
@husigeza Thanks for asking. Yes, this experience was with pymakr (in Atom) and ampy.py

And I have good news: I upgraded the Pycom Upgrade program to 1.16.2, disabled pybytes/smartConfig, and now it worked with 1.20.2.rc3, both with pymakr and ampy.py

closed 2020-02-18 by tschudin