pycom/pycom-micropython-sigfox #403: BLE GATTSCharacteristic change value throws an error

jerom987, 2020-02-19:
Using LoPy4
release=1.20.2.rc3, version=v1.11-2037465

I have following code:
`
import time
import pycom
from network import Bluetooth

bluetooth = Bluetooth()
bluetooth.set_advertisement(name='LoPy', service_uuid=0x1800)

#Handling connection events
def conn_cb (bt_o):
    events = bt_o.events()
    if  events & Bluetooth.CLIENT_CONNECTED:
        print("Client connected")
    elif events & Bluetooth.CLIENT_DISCONNECTED:
        print("Client disconnected")

bluetooth.callback(trigger=Bluetooth.CLIENT_CONNECTED | Bluetooth.CLIENT_DISCONNECTED, handler=conn_cb)

bluetooth.advertise(True)
arg=None

#Handling automation IO service
btsrv_Automation = bluetooth.service(uuid=0x1815, isprimary=True, nbr_chars=1)
btchr_AI1 = btsrv_Automation.characteristic(uuid=0x00012A58, value=1)

def btchr_AI1_cb_handler(chr, arg):
    events = chr.events()
    if  events & Bluetooth.CHAR_READ_EVENT:
        return 'AI 1 VALUE'

btchr_AI1_cb = btchr_AI1.callback(trigger=Bluetooth.CHAR_READ_EVENT, handler=btchr_AI1_cb_handler, arg=arg)

while(True):
    btchr_AI1.value(1234)
    time.sleep(5)`

I get following error when changing the value (by running `btchr_AI1.value(1234)`)
OSError: Erorr while sending BLE indication/notification

mo-norant, 2020-07-02:
anyone?

mate-pycom, 2020-07-02:
Hello @jerom987 and @mo-norant,

The problem was here that when you try to change value of Client, it will send an indication, but in case Client is still not connected, this sending will be failed. I modified code to send indication only if Client is already connected:

https://github.com/pycom/pycom-micropython-sigfox/pull/446/commits/0745b3dd337863f29b05a645ed3de8c7416cc335

I am working on merging this fix into the next release.

Best regards,
Máté

editicalu, 2021-04-21:
Any update on this?

editicalu, 2021-04-24:
For anyone in the future: I worked around this by just `try`-`except`ing the call to the function (and no nothing in the except). As far as I can tell, this shouldn't break anything.