pycom/pycom-micropython-sigfox #406: WLAN stops responding temporarily for 60 seconds with no error

dane-if3, 2020-02-21:
(sysname='WiPy', nodename='WiPy', release='1.20.2.rc3-0.8.0-vanilla-squirrel', version='v1.20.2.rc3-4-g4962aea3b-dirty on 2020-02-11', machine='WiPy with ESP32')

I have a machine (WiPy) with an HMI connected via Modbus via WLAN.
(the machine is the modbus server and the HMI is the modbus client, an iphone in this case)
I can start the machine with out HMI connected and ping it for hours.
I can connect the HMI (or several HMI devices) and have it idle and still ping it, possibly less reliably but still good.
(It ends up with the WiPy unresponsive and the heartbeat light solid blue. Different unresolved bug not filed because I have no idea where to begin)
But when I use the machine I get a strange socket failure where the whole WLAN seams to stall  and pause for a minute or so. I say pause because I have seen it reply to the last modbus request sent before the pause when it comes back on line. The HMI modbus client has  abandoned the connection and been trying to reconnect so it doesn't receive the reply.
During this time the WiPy cant be pinged and doesn't reply to ARP requests etc.
Then it comes back to life most of the time and life goes on.
The rest of the machine continues to operate as normal through out the pause. The sockets are checked via select.poll(5) where nothing is returned.
The problem often occurs a short time after pressing continue on the HMI at a certain stage of the machines operation but not always or at the same place in the code and occasionally not at all. After it has happened once it is less likely to happen again until the system is rebooted.

Below is the output from the main loop, and a ping from the computer attached is the pcap
(I think wireshark sees all the duplicate packets due to the way I captured the Wifi at the router)

I started the capture, ping and booted the machine.
once in the main loop I connected the HMI and moved the machine around selected a preset procedure, initiated a reset when it was ready (Reset complete -> ready) 
I pressed continue (Packet number 3730 in the pcap file)
0.06s later the WLAN stopped responding (before the client can send the continue button released) packet 3742
I monitor the time of the last good data from the modbus socket.
this is the number in the line (Active Connections: [[<socket>, 28074]])
if no good data has been received in 1500ms I close the socket
(Connection closed due to inactivity)
the machine continued on with out the WLAN
77s later when it came back packet 3884
I reconnected the HMI and basically did the same thing a second time.
continue button: pcap 8178
this time it ran for 15 seconds before it did the pause thing (around packet 9743)
paused for 60 seconds and back at packet 9923

i have removed all threads, irqs are enabled because poll/sockets dont work if they are disabled
I have tried to comment out code but it doesn't seem to make much difference as the issue occurs at different places or sometimes not at all
Any suggestions on how to narrow this down?
[packet_capture-11.pcap.zip](https://github.com/pycom/pycom-micropython-sigfox/files/4236221/packet_capture-11.pcap.zip)

`INFO:None: Entering Main Loop
Active Connections: []
New connection Client address:('192.168.xxx.xxx', 55593) socket:<socket>
New state: Stopped
program change id:1
New state: Stopped
Continuing to state -3
New state: Resetting
Actuator enabled for reset
New state: Stopped
Active Connections: [[<socket>, 28074]]
program change id:2
New state: Stopped
Continuing to state -3
New state: Resetting
Actuator enabled for reset
long loop 176. Sensor 124, State 4, PID 1, Modbus 31. Sensor times {'f': 116, 'T2': 2, 'Si': 2, 'T1': 2, 'S': 1} 37034
Reset complete -> ready
Active Connections: [[<socket>, 49817]]
Continuing to state 0
New state: Initialiseing
INFO:None: Heater enabled for testing
New state: Home
Connection closed due to inactivity
long loop 163. Sensor 122, State 0, PID 5, Modbus 5. Sensor times {'f': 3, 'T2': 1, 'Si': 2, 'T1': 114, 'S': 1} 60903
Active Connections: []
New state: Tare with drag
New state: Home
New state: Insert test piece
New state: Wait for temperature
long loop 128. Sensor 9, State 1, PID 111, Modbus 5. Sensor times {'f': 3, 'T2': 1, 'Si': 2, 'T1': 2, 'S': 1} 79921
Active Connections: []
long loop 162. Sensor 121, State 1, PID 4, Modbus 5. Sensor times {'f': 3, 'T2': 1, 'Si': 2, 'T1': 2, 'S': 2} 99030
Active Connections: []
[... 307 lines cut ...]
64 bytes from 192.168.xxx.yyy: icmp_seq=264 ttl=255 time=10.419 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=265 ttl=255 time=8.945 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=266 ttl=255 time=9.958 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=267 ttl=255 time=5.054 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=268 ttl=255 time=9.526 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=269 ttl=255 time=10.644 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=270 ttl=255 time=10.467 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=271 ttl=255 time=6.425 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=272 ttl=255 time=10.861 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=273 ttl=255 time=4.020 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=274 ttl=255 time=16.448 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=275 ttl=255 time=9.256 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=276 ttl=255 time=10.525 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=277 ttl=255 time=10.080 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=278 ttl=255 time=10.306 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=279 ttl=255 time=4.124 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=280 ttl=255 time=8.436 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=281 ttl=255 time=9.219 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=282 ttl=255 time=11.398 ms
64 bytes from 192.168.xxx.yyy: icmp_seq=283 ttl=255 time=9.051 ms
^C
--- 192.168.xxx.yyy ping statistics ---
284 packets transmitted, 153 packets received, 46.1% packet loss
round-trip min/avg/max/stddev = 3.118/47.540/2856.555/280.694 ms
`

robert-hh, 2020-02-21:
I made the same experience of WiFi practically unresponsive, if my Python code would execute in a thigh busy loop. Adding a short time.sleep_ms(delay) imroved the behavior significantly.  The value for delay can be as small as 1. What happens is, that on calling any of the sleep functions, pending tasks are scheduled.

amotl, 2020-02-21:
@robert-hh: Would it also work just using `time.sleep_ms(0)` in order to yield to the scheduler?

robert-hh, 2020-02-21:
@amotl That#s hard to tell. sleep_ms() calls mp_hal_delay_ms() calls vTaskDelay() without checking the value of the argument. So it's up to testing. But idle() may have the same effect, which just calls taskYIELD().

dane-if3, 2020-02-22:
@robert-hh Sorry I didn't include the code for the loop. (see below)
I use sleep_ms to force the loop to run at 50ms or more to be sure the external ADC is ready with a new value.
But even if the last loop was longer than 50ms I still sleep for 1ms (I had sockets in threads and it worked better with several short sleeps. I didn't change this when I swapped the threaded sockets to select.poll() sockets in the main thread)

The main difference before and after pressing continue is a small amount of code calculating PIDs to set the duty cycles for both PWMs and the PWMs are both running
Both PWM can be running with no problem I am not saying they are the issue.

`try:
    while not E_Stop():
        #wdt.feed()
        ### Don't cycle quicker than the minimum loop time.
        t = utime.ticks_ms()
        looptime_period = utime.ticks_diff(t,looptime_last)
        looptime_last = t
        if looptime_period > 2*looptime_min:
            print('long loop {}. Sensor {}, State {}, PID {}, Modbus {}. Sensor times {} {}'.format(looptime_period, sensorUpdateTime,stateUpdateTime,pidUpdateTime,ModbusUpdateTime, test_machine.sensors.ts, t))
        utime.sleep_ms(max(1,looptime_min-looptime_period))

        irq_state=machine.disable_irq()
        ### Update the sensor data
        sensorUpdateTime=-utime.ticks_ms()
        with a_lock:  #print("a_lock is locked while this executes")
            test_machine.sensors.update() # will throw a hard limit error finally can sort it out at the momnet need to handle this proprerly
        sensorUpdateTime+=utime.ticks_ms()

        ### Update/Run the active state
        stateUpdateTime=-utime.ticks_ms()
        if not procedure_manager.runningPocedure.update_state(): break
        stateUpdateTime+=utime.ticks_ms()

        ### Update the controllers
        pidUpdateTime=-utime.ticks_ms()
        test_machine.heater.update()
        test_machine.actuator.update()
        pidUpdateTime+=utime.ticks_ms()
        machine.enable_irq(irq_state)

        ### Output Modbus data
        ModbusUpdateTime=-utime.ticks_ms()
        server.poll(5)
        ModbusUpdateTime+=utime.ticks_ms()

        if i % 500==0:
            print('Active Connections:',server.active_connections)
        i=i+1
`

dane-if3, 2020-02-27:
This is mostly a problem caused by me.
Pycom development boards use pin 12 to switch between the internal and external wifi antennas. 
This can also be done with WLAN.init( ... antenna=WLAN.INT_ANT) which is the only way to do it on the OEM versions of the device. 
On the expansion board Pin 12 is also Pin.exp_board.G28 which is the pin I used for the PWM channel for the heater PID controller.
While the temperature is close to the setpoint the duty cycle is set to zero or close enough to zero to not effect the wifi. If the temperature drops, door opened or machine stopped for a while, the duty cycle increases and the wifi started switching to the external non connected antenna for long enough periods that packets and retransmissions were missed.
So the randomness of this and the G28 pin made it harder to realise I had not read the manual correctly...
It would be nice to have an exception alerting the user that it is a bad idea to use the pin when wifi is enabled. I cant see that this is practical to implement.

closed 2020-02-27 by dane-if3