pycom/pycom-micropython-sigfox #411: Can't send long AT commands using lte.send_at_cmd

LaRox11, 2020-02-25:
* GPy + pytrack or pysense
* (sysname='GPy', nodename='GPy', release='1.20.1.r2', version='v1.11-06dfad0 on 2019-11-30', machine='GPy with ESP32', pybytes='1.3.0')

Hello,

for my application I need to send long AT+CSIM commands (over 500 bytes) to the modem. Using `lte.send_at_cmd` with commands that size causes a crash. I followed the source code through the rabbit hole and found that the buffer behind this method is only 127 bytes big and a carriage return is automatically appended to the command here: https://github.com/pycom/pycom-micropython-sigfox/blob/Release/v1.20/esp32/lte/lteppp.c#L662 which hinders me from sending my command in several chunks.

Is there maybe another way to safely write to the UART without a carriage return being added?

Thank you!

robert-hh, 2020-02-25:
As far as I understand the cited code, the CR is only sent if the command is "+++".

LaRox11, 2020-02-25:
I feel for that, too. It looks counter-intuitive but `strcmp` returns 0 if the contents of both strings are equal and 0 evaluates to `false`.

robert-hh, 2020-02-25:
Oops. You are right.

razvandragomirescu, 2021-11-08:
I am experiencing the same issue, we are sending long command APDUs to the SIM card via AT+CSIM and we run out of space for some commands. This issue is almost 2 years old now, could you please fix it?