pycom/pycom-micropython-sigfox #425: LoRa reset causes a core dump

robert-hh, 2020-03-25:
Calling lora.reset() results in a core dump.
Test code:
```
from network import LoRa
lora = LoRa(mode=LoRa.LORA, frequency=868000000, region=LoRa.EU868)
lora.reset()
```

peter-pycom, 2020-03-25:
Which fw and device are you using? I just tried on FiPy with 1.20.2.rc and couldn't reproduce it. Is it happening sometimes or always?

robert-hh, 2020-03-26:
LoPy4 with 1.20.2.rc6 fails
FiPy with the same version is stable

mo-norant, 2020-08-14:
Could somebody please elaborate? @peter-pycom

robert-hh, 2020-08-15:
The error is still present with 1.20.2.r0. Below is a detailed core dump analysis:
```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffd26dc, name: 'LoRa', GDB name: 'process 1073555164'

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x14
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: value_of_register_lazy: Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: value_of_register_lazy: Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
Reading symbols from ../esp32/build/LOPY4/release/application.elf...done.
[New process 1073555164]
[New process 1073468752]
[New process 1073461568]
[New process 1073462964]
[New process 1073455028]
[New process 1073562564]
[New process 1073466144]
[New process 1073469464]
[New process 1073597508]
[New process 1073558608]
[New process 1073445480]
[New process 1073445936]
[New process 1073459364]
[New process 1073443452]
[New process 1073431364]
[New process 1073497564]
[New process 1073459996]
[Current thread is 1 (process 1073555164)]

==================== CURRENT THREAD STACK =====================
#-1 0x00000000 in ?? ()
Backtrace stopped: Cannot access memory at address 0x0

======================== THREADS INFO =========================
  Id   Target Id         Frame 
  17   process 1073459996 0x4000bff0 in ?? ()
  16   process 1073497564 xEventGroupWaitBits (xEventGroup=0x3ffdffc8, uxBitsToWaitFor=3, xClearOnExit=1, xWaitForAllBits=0, xTicksToWait=<optimized out>) at /Users/ehlers/pycom/pycom-esp-idf/components/freertos/event_groups.c:445
  15   process 1073431364 0x40095e54 in xQueueGenericReceive (xQueue=0x3ffb2e50, pvBuffer=0x3ffb42a0, xTicksToWait=4294967295, xJustPeeking=0) at /Users/ehlers/pycom/pycom-esp-idf/components/freertos/queue.c:1592
  14   process 1073443452 0x40095e54 in xQueueGenericReceive (xQueue=0x3ffc4030 <s_timer_semaphore_memory>, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /Users/ehlers/pycom/pycom-esp-idf/components/freertos/queue.c:1592
  13   process 1073459364 0x40095e54 in xQueueGenericReceive (xQueue=0x3ffd86c4, pvBuffer=0x3ffd9860, xTicksToWait=4294967295, xJustPeeking=0) at /Users/ehlers/pycom/pycom-esp-idf/components/freertos/queue.c:1592
  12   process 1073445936 0x40095e54 in xQueueGenericReceive (xQueue=0x3ffb7bd8, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /Users/ehlers/pycom/pycom-esp-idf/components/freertos/queue.c:1592
  11   process 1073445480 0x40095e54 in xQueueGenericReceive (xQueue=0x3ffb760c, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /Users/ehlers/pycom/pycom-esp-idf/components/freertos/queue.c:1592
  10   process 1073558608 0x4000bff0 in ?? ()
  9    process 1073597508 0x40095e54 in xQueueGenericReceive (xQueue=0x3ffbd6c0, pvBuffer=0x3ffded00, xTicksToWait=4294967295, xJustPeeking=0) at /Users/ehlers/pycom/pycom-esp-idf/components/freertos/queue.c:1592
  8    process 1073469464 0x4000bff0 in ?? ()
  7    process 1073466144 /builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
[... 135 lines cut ...]
.coredump.tasks.data 0x3ffbd818 0x16c RW 
.coredump.tasks.data 0x3ffdc790 0x194 RW 
.coredump.tasks.data 0x3ffdcc44 0x16c RW 
.coredump.tasks.data 0x3ffdeba0 0x210 RW 
.coredump.tasks.data 0x3ffd3450 0x16c RW 
.coredump.tasks.data 0x3ffd3280 0x1c8 RW 
.coredump.tasks.data 0x3ffb7a68 0x16c RW 
.coredump.tasks.data 0x3ffb78e0 0x180 RW 
.coredump.tasks.data 0x3ffb7c30 0x16c RW 
.coredump.tasks.data 0x3ffb9cb0 0x188 RW 
.coredump.tasks.data 0x3ffbb0a4 0x16c RW 
.coredump.tasks.data 0x3ffd9760 0x1b8 RW 
.coredump.tasks.data 0x3ffb727c 0x16c RW 
.coredump.tasks.data 0x3ffb70e0 0x194 RW 
.coredump.tasks.data 0x3ffb4344 0x16c RW 
.coredump.tasks.data 0x3ffb4170 0x1cc RW 
.coredump.tasks.data 0x3ffc45dc 0x16c RW 
.coredump.tasks.data 0x3ffd7fd0 0x6ec RW 
.coredump.tasks.data 0x3ffbb31c 0x16c RW 
.coredump.tasks.data 0x3ffda780 0x19c RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

robert-hh, 2020-08-15:
OK. I found the culprit for the crash. TASKLora calls in line 809 of esp32/mods/modlora.c Radio.Reset(). But that entry is not initialized. The code to do that is missing in esp32/lora/sxx1276-board.c. Line 53 of the file has to be replaced by:
```
    SX1276SetPublicNetwork,
    SX1276Reset
```
That adds the missing entry. And in drivers/sx127x/sx1276/sx1276.h the declaration of SX1276Reset has to be added, like:
```
/*!
 * \brief Resets the SX1276
 */
void SX1276Reset( void );
```
After making these changes the board does not crash any more. Whether it resets has to be told.

robert-hh, 2023-07-17:
Closing because Pycom was closed.

closed 2023-11-14 by robert-hh