pycom/pycom-micropython-sigfox #430: Gpy dies after update with 1.20.2.rc6, must downgrade to r2

DG12, 2020-04-08:
Gpy was running a boot.py which (among other things) set LED blue, set WLAN to STAtion and connected, then ran main.py to get ntp server time(among other things).

After updating to 1.20.2.rc6 main executes as indicated by blue LED but can no longer be accessed by either telnet or ftp from lan as station.
Erasing and flash again I expected Gpy to enter AccessPoint(since there is no longer a boot.py to set station) but it only transmits the SSID once and a MacBook Pro can no longer connect. The same problem exists with version r3.

Flash to version r2 and it is now accessible as AP, ftp boot.py and main.py and operation continues as before update.

More detail can be found at forum discussion [new-firmware-release-1-20-2-rc6/9](https://forum.pycom.io/topic/5740/new-firmware-release-1-20-2-rc6/9)

robert-hh, 2020-05-07:
This may be caused by the pybytes firmware & mechanisms. Try to use a non-pybytes version. Pycom calls that legacy and only offers old releases. There is also a set of files maintained by @andreas which do not include Pybytes and are recent. They also do avoid some of the bugs Pycom did not fix yet, like to the littlefs file system.
https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/

DG12, 2020-05-07:
@robert-hh Thanks for your reply. I don't use pybytes. I connect via WLAN and ftp source to the Gpy. 

I noticed the descriptions for hiveeyes vanilla state that it uses littleFS which I thought you weren't comfortable with.

robert-hh, 2020-05-08:
With the changes to LittleFs, which both me and @amotl made, it runs much more stable. It did not crash since then.

peter-pycom, 2020-05-08:
@robert-hh are you referring to these changes or something else? https://github.com/pycom/pycom-micropython-sigfox/pull/418

robert-hh, 2020-05-08:
Some of them. As I have written in my comments, I do not agree to all.
The other aspect I do not like is the attempt to enforce pybytes. Even if newer builds disable it by default, it is then quite a piece of dead code. 
Besides that I honour much the improvements made, especially of the documentation.