pycom/pycom-micropython-sigfox #431: After CTRL F, Gpy closes all ports but does not reset the board requires manual reset

DG12, 2020-04-08:
`(sysname='GPy', nodename='GPy', release='1.20.2.rc2', 

version='v1.11-605f40f on 2020-02-04', machine='GPy with ESP32', pybytes='1.3.1')`

Connecting via telnet and entering CTRL F on a new line closes all ports, telnet reports Connection closed by foreign host. ping reports "ping: sendto: Host is down" and ftp times out.

Pressing the reset button is the only way to recover.

robert-hh, 2020-04-08:
That's an intended behavior, a so-called feature. Pushing Ctrl-F in REPL causes a safe-boot, which is permanent until the reset button is pressed or machine.reset() is called. When that was introduced first, it caused me some trouble, because it could not be disabled. Calling micropython.kbd_intr(-1) disables this special behavior of both Ctrl-F and Ctr-C, again until restart. A PR I made to disable Ctrl-F and Ctrl-C separately was not implemented.

peter-pycom, 2020-04-08:
As robert said, this is intentional. Ctrl-F puts it in safe-boot

robert-hh, 2020-04-08:
Even if there is a mechanism to disable safe-boot after Ctrl-F, this is not implemented for Telnet. Telnet still checks for a hard-coded CHAR_CTRL_F constant,  instead of referring to mp_reset_char. I had implemented the latter in my PR, but it was not pulled in by Pycom, whether intentionally or as omission. I asked a few times w/o feedback.

peter-pycom, 2020-04-08:
I can't find the PR. Can you point me to it?

robert-hh, 2020-04-08:
That's quite a while ago. I make a new one.

DG12, 2020-04-08:
Perhaps a clarification in the help and documentation would be helpful since the system does not boot when issuing a CTRL-F

robert-hh, 2020-04-08:
I see what you mention. Push Ctrl-F in Telnet causes the device to freeze. I would consider that a bug. So the question is, whether response to Ctrl-F should not be removed at all in telnet. The purpose of Ctrl-F is to enforce a well defined counterpart when used with Pymakr & VSC/Atom.
b.t.w.: The response to Ctrl-D in telnet is weird as well.

DG12, 2020-04-08:
I noticed that CTRL-D also does not work as ( I ) expected.

Perhaps reopen this issue?

robert-hh, 2020-04-08:
To have Ctrl-F result in a (safe) reboot is easy. Just change in line 541 false -> true. The Ctrl-D stuff is more weird. Because, when Telnet is opened, Ctrl-D will not result in a reboot, until enter is pressed in the telnet session; even when Ctrl-D was pushed in the UART session.
OK. found a brute-force fix, which closes the telnet session upon Ctrl-D and reboots.

robert-hh, 2020-04-08:
OK. The PR #432 is updated.