pycom/pycom-micropython-sigfox #434: Many Core Dumps when wlan in ON!

ElinhoHD, 2020-04-20:
The Pycom board is a W01 (Wipy version 1.20.1.r2 (Micropython 1.11 , ESP32 v 3.4.0)

```
(sysname='WiPy', nodename='WiPy', release='1.20.1.r2', version='v1.11-06dfad0 on 2019-11-30', machine='WiPy with ESP32', pybytes='1.3.0')
```
Hi, Since I use LittleFS on my project and let the wlan running, I have the problem that the component restart by itself leaving different messages like this (from ESP32):
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40096e7f  PS      : 0x00060f33  A0      : 0x80094973  A1      : 0x3ffdf360
A2      : 0x3ffb7730  A3      : 0x00060f20  A4      : 0x3ffd4794  A5      : 0x00000001
A6      : 0x000000fe  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffc548c
A10     : 0x3ffc548c  A11     : 0x00000000  A12     : 0x3ffd4794  A13     : 0x00000001
A14     : 0x000000fe  A15     : 0x3ffc54f0  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4009bd80  LEND    : 0x4009bdae  LCOUNT  : 0xffffffff

Backtrace: 0x40096e7f:0x3ffdf360 0x40094970:0x3ffdf380 0x400eed08:0x3ffdf3a0 0x400de3fd:0x3ffdf430
```
If you need some details about the project I am available on this email address: mrntumba@gmail.com
Thank you!

peter-pycom, 2020-04-20:
Is it also happening with FatFS?

ElinhoHD, 2020-04-20:
No but when I using FatFS, I have another problem. My device is losing datas.

ElinhoHD, 2020-04-21:
> Is it also happening with FatFS?
@peter-pycom
No but when I using FatFS, I have another problem. My device is losing datas.

amotl, 2020-04-21:
Dear @ElinhoHD,

with our Squirrel builds [1], errors like you've outlined might just go away.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960

ElinhoHD, 2020-04-22:
Hello, I also use the squirell builds(WiPy-1.20.2.rc6-0.10.1-vanilla-squirrel.tar.gz) 
When I m using my WLAN I have a core dump and a 
total machine standstill.

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40096ef3  PS      : 0x00060a33  A0      : 0x800953c3  A1      : 0x3ffdcb70      
A2      : 0x3ffe3d18  A3      : 0x00060a20  A4      : 0x3ffdab58  A5      : 0x00000001      
A6      : 0x000000fe  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffc3fc0      
A10     : 0x3ffc3fc0  A11     : 0x0000abab  A12     : 0x3ffdab58  A13     : 0x00000001      
A14     : 0x000000fe  A15     : 0x3ffe3d10  SAR     : 0x00000000  EXCCAUSE: 0x0000001c      
EXCVADDR: 0x00000004  LBEG    : 0x40091d54  LEND    : 0x40091d82  LCOUNT  : 0xffffffff      

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40096ef3:0x3ffdcb70 0x400953c0:0x3ffdcb90 0x400f3aa4:0x3ffdcbb0 0x400e05d9:0x3ffdcc40
```

ElinhoHD, 2020-04-22:
@amotl

amstaff8, 2020-10-05:
Hi @ElinhoHD 

Did you try to decrease the power transmission of WIFI?

I had a similar problem and it helped me: https://forum.pycom.io/topic/6389/w01-crashes-with-wifi-tx-at-max-power/2

However I would like to know why the W01 behaves like this.