pycom/pycom-micropython-sigfox #437: Hi everyone i'm using a development board lopy4 and i wanted to have the PCB design to know the capacities and resistors that are linked to the Pins i have founded the pcb design for the universal OEM baseboard is it the same

leila111111, 2020-04-28:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
import os
os.uname()
```
The output will look similar to this:
```
(sysname='LoPy', nodename='LoPy', release='0.9.4.b1', version='bc42ea7 on 2016-11-14', machine='LoPy with ESP32')
```
* Exact steps to cause this issue
    1. First I did this...
    2. Then, I did that...
* What you expected
    * e.g. I expected *abc* to *xyz*
* What happened instead
    * e.g. Instead, I got *zyx*

Thank you!

dmpycom, 2020-05-04:
Hello leila111111!

Please refer LoPy4 datasheet minimum required circuit on page 27: https://docs.pycom.io/gitbook/assets/specsheets/Pycom_002_Specsheets_LoPy4_v2.pdf 

As shown in the minimum required circuit decoupling capacitors are not required. In case if power supply wires too long, you can place close to power pins a ceramic or electrolytic capacitor in range 10uF .. 470uF

OEM modules require external voltage regulator and decoupling capacitors

leila111111, 2020-05-19:
Thanks for your reply.