pycom/pycom-micropython-sigfox #443: Core dump during FTP file transfer, caused by a heap lock.

robert-hh, 2020-05-10:
Pycom MicroPython 1.20.2.rc7 [v1.20.1.r2-142-g6d01270-dirty] on 2020-05-10; LoPy4 with ESP32

This is the "a smoking gun" for a fault, which made me hate LittleFS. It happens when I do other activities in REPL or an active Python script during or shortly after making FTP file transfers. Replacing m_malloc() by malloc() in lfs_util.h solved the problem, and now I found the reason, different to my initial assumption. Below is the backtrace and the core dump. The reason is, that lfs tries to allocate memory for a buffer, but the heap is locked at that very moment. Since malloc() does not use the heap, the problem does not occur.
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400f66da  PS      : 0x00060c30  A0      : 0x800fe6fe  A1      : 0x3ffdba80  
A2      : 0x3ffc93ec  A3      : 0x3f40d2df  A4      : 0x3ffc93d4  A5      : 0x00000008  
A6      : 0x20200408  A7      : 0x0000042f  A8      : 0x800f66da  A9      : 0x3ffdba60  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3f40f684  A13     : 0x00000000  
A14     : 0x3f40e08c  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x40093d79  LEND    : 0x40093d89  LCOUNT  : 0xfffffffd  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400f66da:0x3ffdba80 0x400fe6fb:0x3ffdbaa0 0x400fe712:0x3ffdbac0 0x400f6712:0x3ffdbae0 0x400f50bc:0x3ffdbb00 0x400f5324:0x3ffdbb30 0x400f6006:0x3ffdbfa0 0x400e0240:0x3ffdbfd0

Simple backtrace:

BT-0: 0x400f66da is in nlr_jump (../py/nlrsetjmp.c:33).
33	    nlr_buf_t *top = *top_ptr;
BT-1: 0x400fe6fb is in mp_raise_msg (../py/runtime.c:1465).
1465	        nlr_raise(mp_obj_new_exception_msg(exc_type, msg));
BT-2: 0x400fe712 is in m_malloc_fail (../py/runtime.c:1454).
1454	        mp_raise_msg(&mp_type_MemoryError, "memory allocation failed, heap is locked");
BT-3: 0x400f6712 is in m_malloc (../py/malloc.c:88).
88	        m_malloc_fail(num_bytes);
BT-4: 0x400f50bc is in f_readdir_helper (ftp/ftp.c:341).
341	                char* file_relative_path = m_malloc(path_length);
BT-5: 0x400f5324 is in ftp_list_dir (ftp/ftp.c:1455).
1455	            f_readdir_helper(&ftp_data.u.dp, &fno);
BT-6: 0x400f6006 is in ftp_run (ftp/ftp.c:672).
672	                ftp_list_dir((char *)ftp_data.dBuffer, FTP_BUFFER_SIZE, &listsize);
BT-7: 0x400e0240 is in TASK_Servers (serverstask.c:105).
105	            ftp_run();

Core dump analysis:
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbd550, name: 'Servers', GDB name: 'process 1073468752'

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x1c
epc7           0x10
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x400f66da	0x400f66da <nlr_jump+6>
lbeg           0x40093d79	1074347385
lend           0x40093d89	1074347401
lcount         0xfffffffd	4294967293
sar            0x18	24
ps             0x60c20	396320
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
[... 200 lines cut ...]
.coredump.tasks.data 0x3ffd42f4 0x16c RW 
.coredump.tasks.data 0x3ffd40e0 0x20c RW 
.coredump.tasks.data 0x3ffbcb20 0x16c RW 
.coredump.tasks.data 0x3ffbc9c0 0x158 RW 
.coredump.tasks.data 0x3ffbd818 0x16c RW 
.coredump.tasks.data 0x3ffdcec0 0x194 RW 
.coredump.tasks.data 0x3ffd3380 0x16c RW 
.coredump.tasks.data 0x3ffd31b0 0x1c8 RW 
.coredump.tasks.data 0x3ffb7c30 0x16c RW 
.coredump.tasks.data 0x3ffb9c90 0x1a8 RW 
.coredump.tasks.data 0x3ffbb13c 0x16c RW 
.coredump.tasks.data 0x3ffd9690 0x1b8 RW 
.coredump.tasks.data 0x3ffdd374 0x16c RW 
.coredump.tasks.data 0x3ffdf2d0 0x210 RW 
.coredump.tasks.data 0x3ffb7a68 0x16c RW 
.coredump.tasks.data 0x3ffb78e0 0x180 RW 
.coredump.tasks.data 0x3ffb4344 0x16c RW 
.coredump.tasks.data 0x3ffb4170 0x1cc RW 
.coredump.tasks.data 0x3ffb727c 0x16c RW 
.coredump.tasks.data 0x3ffb70e0 0x194 RW 
.coredump.tasks.data 0x3ffbb35c 0x16c RW 
.coredump.tasks.data 0x3ffda6b0 0x19c RW 

===================== ESP32 CORE DUMP END =====================
```

peter-pycom, 2020-05-11:
@robert-hh thanks for the detailed analysis and thanks for your (hopefully) patience. The GC alloc problem is being worked on right now. @salal-m please take this repro into account!

salal-m, 2020-05-14:
@robert-hh we have incorporated your suggestions with our changes. These solve the memory related crashes but I was not able to produce the scenario where heap is locked. Could you please verify if the attached dev build is stable in your scenario as well? Thanks.
[LoPy4-1.20.2.rc7.tar.gz](https://github.com/pycom/pycom-micropython-sigfox/files/4627376/LoPy4-1.20.2.rc7.tar.gz)

robert-hh, 2020-05-14:
Yes, this is hard to trigger, when you want to. I will test it. I do not know what exactly you changed.  Below is the function as I have it now, using malloc. At least for the m_malloc() exception, it should be possible to use it, when before the call a return frame is pushed with nlr_push() (and of course later removed with nlr_pop(). But looking at all other places in ftp.c where memory is allocated, it is consistent NOT to use m_malloc().
```
STATIC FRESULT f_readdir_helper(ftp_dir_t *dp, ftp_fileinfo_t *fno ) {

    if(isLittleFs(ftp_path))
    {

        vfs_lfs_struct_t* littlefs = lookup_path_littlefs(ftp_path, &path_relative);
        if (littlefs == NULL) {
            return FR_NO_PATH;
        }

        xSemaphoreTake(littlefs->mutex, portMAX_DELAY);

            int lfs_ret = lfs_dir_read(&littlefs->lfs, &dp->u.dp_lfs, &fno->u.fpinfo_lfs.info);

            // Fetch the timestamp for valid files, filter out empty, "." and ".." entries
            if((lfs_ret >= LFS_ERR_OK) &&
               !(fno->u.fpinfo_lfs.info.name[0] == 0) &&
               !(fno->u.fpinfo_lfs.info.name[0] == '.' && fno->u.fpinfo_lfs.info.name[1] == 0) &&
               !(fno->u.fpinfo_lfs.info.name[0] == '.' && fno->u.fpinfo_lfs.info.name[1] == '.' && fno->u.fpinfo_lfs.info.name[2] == 0))
            {
                // Length of the relative path
                size_t length_of_relative_path = strlen(path_relative);

                // Calculate the length of the relative path of the file with the name of the file plus closing /0
                size_t path_length = length_of_relative_path + strlen(fno->u.fpinfo_lfs.info.name) + 1;

                // If the current directory is not root directory need to append extra "/" at the end of relative path
                if(length_of_relative_path > 1) {
                    path_length++;
                }
                char* file_relative_path = malloc(path_length);

                if (file_relative_path) {

                    // Copy the current working directory (relative path)
                    memcpy(file_relative_path, path_relative, length_of_relative_path);

                    // Append the "/" at the end of current working directory path if needed
                    if(length_of_relative_path > 1) {
                        memcpy(&file_relative_path[length_of_relative_path], "/", 1);
                        // Modify the length of relative path to include the closing "/"
                        length_of_relative_path++;
                    }
                    // Copy the name of the file after the current working directory, this will copy the closing /0
                    strcpy(&file_relative_path[length_of_relative_path], fno->u.fpinfo_lfs.info.name);

                    int lfs_getattr_ret = lfs_getattr(&littlefs->lfs, file_relative_path, LFS_ATTRIBUTE_TIMESTAMP, &fno->u.fpinfo_lfs.timestamp, sizeof(lfs_timestamp_attribute_t));
                    // If no timestamp is saved for this entry, fill it with 0
                    if(lfs_getattr_ret < LFS_ERR_OK) {
                        fno->u.fpinfo_lfs.timestamp.fdate = 0;
                        fno->u.fpinfo_lfs.timestamp.ftime = 0;
                    }

                    free(file_relative_path);
                } else {
                    lfs_ret = LFS_ERR_IO;
                }
            }

        xSemaphoreGive(littlefs->mutex);

        return lfsErrorToFatFsError(lfs_ret);
    }
    else
    {
        FATFS *fs = lookup_path_fatfs(ftp_path, &path_relative);
        if (fs == NULL) {
            return FR_NO_PATH;
        }
        return f_readdir(&dp->u.dp_fat, &fno->u.fpinfo_fat);
    }
}
```

salal-m, 2020-05-14:
You are right, m_malloc() should not be used here. Even if the crash doesn't occur due to locked heap, there are instances of crashes due to invalid memory access (which most likely happens because GC collects that memory).
The changes mostly include memory allocations replaced by _malloc()_ in all modules where it is to be used internally for non-MircoPython use. (including instances in PR #418)

robert-hh, 2020-05-14:
Dang. I was optimistic, since I ran my old test for an hour and it worked fine. Then I started a different test, and that crashed. Since I do not have the elf file of your image, I cannot decode the core dump. So you just get the numbers. The test setup is simple. Use Filezilla to upload a bunch of files. In my case it were ~100 of different sizes. Then I run in REPL, connected through USB/UART:
```
import gc
import uos
gc.threshold(500000)   # just to enforce a few more gc's
while True:
    print (list(uos.ilistdir()))
```
Core dump:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4020f817  PS      : 0x00060630  A0      : 0x800f1de6  A1      : 0x3ffdb660  
A2      : 0x00010009  A3      : 0x3ffc9310  A4      : 0x3ffc47c0  A5      : 0xffffffff  
A6      : 0x00001000  A7      : 0x00000005  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x3ffe6638  A15     : 0x3ffe6638  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00010009  LBEG    : 0x40093c20  LEND    : 0x40093c2b  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4020f817:0x3ffdb660 0x400f1de3:0x3ffdb680 0x400f2584:0x3ffdb700 0x400f25d5:0x3ffdb740 0x400f370a:0x3ffdb760 0x400f52ca:0x3ffdb780 0x400f61e8:0x3ffdb7a0 0x400e0218:0x3ffdb7d0

================= CORE DUMP START =================
9D8AAAEAAAARAAAAbAEAAA==
sNX7P6C1/T9QuP0/
ALX9P/C3/T/8f0IA6EL8P+hC/D+w1fs/4EL8Pw4AAADgdfs/4HX7P7DV+z8AAAAA
CwAAAFSo/T9TZXJ2ZXJzAIzY+z8Q1vsAAQAAAFC4/T8AAAAAIAoGAAsAAAABAAAA
CEf7PwAAAAAAAAAAAAAAAIAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
XYdAPwAAAADwMQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEhBTg==
/////xf4IEAwBgYA5h0PgGC2/T8JAAEAEJP8P8BH/D//////ABAAAAUAAAABAAAA
AAAAAAAAAAABAAAAAAAAAAEAAAA4Zv4/OGb+PxgAAAAcAAAACQABACA8CUArPAlA
AAAAAAAAAAABAAAAzDMIQCyj+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAOskD4CAtv0/wEf8PwiT/D+HJQ+AgLb9PwAAAAAQk/w/
wEf8P/////8AEAAABQAAANglD4AAt/0/iDr+PwiT/D8AAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA/v///wAAAAABAAAATQIAAAAAAAAAAAAA
KwMAAAAAAAAAEAAAyHH8PwAAAACwVgIAALf9P+R+CUAAABAAAgAAAMBH/D8BAAEA
DTcPgEC3/T/AR/w/CJP8PwgMICAQt/0/AQwgEIg6/j9BAQAATQIAAP4AAAAAAAAA
wEf8PxCT/D8BAAAAAAAAAM1SD4Bgt/0/wEf8PwiT/D+w1fs/AQAAAP4AAAAvAAAA
62EPgIC3/T/AR/w/AJP8P1yT/D9ok/w/AAAAAAAAAAAbAg6AoLf9PwCT/D8AAAAA
AAIAAAAAAAD+AAAAAAAAAAAAAADQt/0/MPj7PwAAAAAAAAAA0Lf9PwAAAAAAAAAA
gAAAAMBH/D8BAAAAAAAA/wAAAADwt/0/AAAAAAAAAAAAAAAAAQAAAEJK/D8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA/Lf9PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAA==
DCb9PzAk/T8EJv0/
MCT9P6Al/T+6ckIAhEL8P4RC/D8MJv0/fEL8PxMAAABgFN2bhY8LCQwm/T8AAAAA
BgAAAAgW/T9Mb1JhABkdD0txNpOt4IIAAQAAAAQm/T8AAAAAIQAGAAYAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
XYdAPwAAAADwMQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACJ38g==
MDQIQGwkEUAwCgYAAAAAAPAk/T+Ymfw/LJr8P6g1QT8AAAAALJr8PwAAAAAHLhGA
0CT9PwAAAAAsmvw/GCH+PwAAAAAAAAAAAAAAAAAAAADcEPo/AAAAAAAAAAAAAAAA
AAAAADA2CEAAAAAA5H4JQNwQ+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAABgh/j8AAAAAAAAAAAAAAAAAAAAAoCX9PwAAAAAAAAAA
oCX9PwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAMDYIQAAAAADkfglA
3BD6PwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
qDVBPwAAAAAsmvw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKwl/T8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
QLn7P6C3+z84ufs/
oLf7P9C4+z+UCZVyvL77PwxC/D9Aufs/BEL8PxkAAABnve8TDYAKOkC5+z8AAAAA
[... 300 lines cut ...]
gIX9PwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAAAAADA8GQAAAAAA////////////////
/////72WflgAAAAAAAAAAAAAAAAA8B8AwIP9PwAAAAAAAAAAAAAAAAAAAAAAAAAA
fhdAPwAAKAAAAAAABhdAPwAAAAAAAAAAAAAAAAAAAAABAAAAAAAAAIyF/T8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
HLP7P7Cm/T9MqP0/
sKb9P+Cn/T9gs/s/gNj7P4hB/D8cs/s/gEH8PxQAAAAAAAAAAAAAAByz+z8AAAAA
BQAAAFCY/T9TbWFydENvbmZpZwAAAAAAAQAAAEyo/T8AAAAAIQAGAAUAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
XYdAPwAAAADwMQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAAAAAA==
MDQIQPC/AEAwAAYAIGcJgHCn/T8AAAAAIwAGACAABgDg8Ps/AwAAACMABgD//z+z
q6sAAOzc+z8jAAYAHLP7PwEAAAD+AAAAAAAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAADA2CEABAAAA5H4JQByT+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAmeAmAgKf9P0Td+z8FAAAA
u20OgKCn/T8BAAAA/EP8Pxyz+z8BAAAA/gAAACAAAAAAAAAAwKf9PwAAAAAAAAAA
+EP8P0Td+z9sQfw/cEL8PwAAAADgp/0/AAAAAAAAAAAcT/w/AAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA7Kf9PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
================= CORE DUMP END =================

salal-m, 2020-05-14:
The decoded stack from this is:

==================== CURRENT THREAD STACK =====================
#0  lfs_pair_cmp (paira=0x10009, pairb=0x3ffc9310 <ftp_data+16>) at littlefs/lfs.c:240
#1  0x400f1de6 in lfs_dir_commit (lfs=0x10009, dir=0x3ffc9310 <ftp_data+16>, attrs=0x3ffdb700, attrcount=1073591904) at littlefs/lfs.c:1640
#2  0x400f2587 in lfs_file_sync (lfs=0x3ffc47c0 <sflash_vfs_flash+56>, file=0x3ffc9308 <ftp_data+8>) at littlefs/lfs.c:2595
#3  0x400f25d8 in lfs_file_close (lfs=0x3ffc47c0 <sflash_vfs_flash+56>, file=0x3ffc9308 <ftp_data+8>) at littlefs/lfs.c:2397
#4  0x400f370d in littlefs_close_common_helper (lfs=0x3ffc47c0 <sflash_vfs_flash+56>, fp=0x3ffc9308 <ftp_data+8>, cfg=0x3ffc935c <ftp_data+92>, timestamp_update_ptr=0x3ffc9368 <ftp_data+104>) at littlefs/vfs_littlefs.c:326
#5  0x400f52cd in f_closefile_helper (fp=0x3ffc9308 <ftp_data+8>) at ftp/ftp.c:416
#6  ftp_close_files () at ftp/ftp.c:1257
#7  0x400f61eb in ftp_run () at ftp/ftp.c:733
#8  0x400e021b in TASK_Servers (pvParameters=<optimized out>) at serverstask.c:105

The .elf file is:
[application.elf.tar.gz](https://github.com/pycom/pycom-micropython-sigfox/files/4629129/application.elf.tar.gz)

So, the crash is happening at another point now. For some reason I am not able to reproduce this one either on LOPY4. I transfer ~120 small files (average size 1KB) and then run the above code through REPL. It keeps printing the filenames without any issue. I even tried doing FTP transfer while this was running, still no crash. Does the crash happen immediately and is it consistent?

robert-hh, 2020-05-14:
I will repeat the tests. And you have to do ftp file transfers while the listdirloops

robert-hh, 2020-05-14:
One question about the changes: did you also change the definition of lfs_malloc in lfs_util.h? There m_malloc() was used, and lfs uses that for the allocation of a per-file cache buffer.

salal-m, 2020-05-14:
Yes. That has also been changed to use malloc().

robert-hh, 2020-05-14:
In the absence of your elf file, I had repeated the test with my local build, and could not produce the error again. Now it is running with your firmware again, and stable until now. 
I hope the LoPy4 flash chip will last. But if it fails, I have spare flash chips here.

robert-hh, 2020-05-14:
Looking at the decoded stack from teh core dump, things are definitely wrong. The code compares data, and both pointer should point to RAM addresses. But one of them does not, for whatever reason.

salal-m, 2020-05-15:
Were you able to reproduce this issue?
If the stack is corrupted and this happens while FTP and ilistdir() are running concurrently, then it could mean that one process is doing something to the other's stack somehow.
Could you also perform this test on your build and see if this happens there as well?

robert-hh, 2020-05-15:
I trioed quite a while with both firmware versions, but could not replicate it. That's kind of sad. But a core dump which happens repeatable but only once in  a while is not much better.
 I had seen that it happens while ftp is closing a file it had written to in the ftp thread. So likely a stack problem, maybe stack overflow.

amotl, 2020-05-15:
Hi there,

I'm glad to see you are working on this topic and will be happy to review the current state against #418. We are using FTP intensively with our Squirrel builds [1] and have not seen any issues so far.

@salal-m: May I humbly ask which toolchain you are using to build the firmware images?

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/squirrel-firmware-for-pycom-esp32/2960

robert-hh, 2020-05-15:
I should not that all kudos on this topic go to @amotl, who was the first to mention that and his work for fixing it. After we talked about it, I started to look for the malloc() uses and changed it at two places, ftp.c and lfs_util.h. @amotl male later his PR https://github.com/pycom/pycom-micropython-sigfox/pull/418 with a more extensive list of potentially  affected places.

closed 2020-05-15 by robert-hh