pycom/pycom-micropython-sigfox #450: Hi everyone i'm using the OEM Module L01 and i wanted  to integrate it into my PCB that's why i have to know the capacities and resistors and others components that should be connected to my OEM module but i didn't find it in the datasheet any ideas???? please i need your help

leila111111, 2020-05-19:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
import os
os.uname()
```
The output will look similar to this:
```
(sysname='LoPy', nodename='LoPy', release='0.9.4.b1', version='bc42ea7 on 2016-11-14', machine='LoPy with ESP32')
```
* Exact steps to cause this issue
    1. First I did this...
    2. Then, I did that...
* What you expected
    * e.g. I expected *abc* to *xyz*
* What happened instead
    * e.g. Instead, I got *zyx*

Thank you!

catalinio, 2020-05-26:
Hi, 

Could you look into the schematics of Universal OEM reference board: https://docs.pycom.io/datasheets/oem/universal_reference/?

Thanks,
Catalin

leila111111, 2020-05-26:
okay thank you .