pycom/pycom-micropython-sigfox #451: ssl.wrap_socket [Errno -1] ERR_MEM

tyrant666, 2020-05-22:
(sysname='LoPy4', nodename='LoPy4', release='1.20.2.rc7', version='v1.11-6d01270 on 2020-05-04', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.4.0')

I'm trying to make a secure connection between L04(server) and a website(client). During the ssl module the memory error is displayed. the generated certificate has the following settings (x509 sha256 rsa: 2048) the error is [Errno -1] ERR_MEM.
how can I solve?
Thanks

catalinio, 2020-05-26:
Hi, 
Could you try the build provided in this comment: https://github.com/pycom/pycom-micropython-sigfox/issues/443#issuecomment-628524787?
We're working on a release which can fix multiple memory errors.

tyrant666, 2020-05-26:
Hello, thanks for the feedback I will work on this attempt this week!

tyrant666, 2020-05-26:
tested, same error

geza-pycom, 2020-06-07:
Could you please copy here the parameters of ssl.wrap_socket() ? What is the value of cert_reqs parameter ?