pycom/pycom-micropython-sigfox #459: utime.mktime() is wrong by 2*utime.timezone()

dane-if3, 2020-06-26:
(sysname='WiPy', nodename='WiPy', release='1.20.2.rc7', version='v1.11-6d01270 on 2020-05-04', machine='WiPy with ESP32', pybytes='1.4.0')

utime.mktime(X) returns an incorrect value as per https://docs.pycom.io/firmwareapi/micropython/utime/
the value is x+2*timezone

```python
import utime
utime.timezone(1*60**2) #+1 hours

print('utc+timezone = localtime->the following should be 0')
print(utime.time()+utime.timezone() - utime.mktime(utime.localtime()))

print('utime.time() is the inverse of utime.gmtime() -> the following is equal')
print(utime.gmtime(),utime.gmtime(utime.time()))

print('utime.mktime() is the inverse of utime.localtime() -> the following should be equal')
print(utime.localtime(),utime.localtime(utime.mktime(utime.localtime())))

print('This should be 0 but it is twice the timezone')
print(utime.mktime(utime.localtime(0)))

print('This should be -timezone but it is +timezone or twice timezone from the correct value')
print(utime.mktime(utime.gmtime(0)))

print('The following should be stable values not increasing')
print('local')
t=utime.localtime()
for i in range(5):
    t=utime.localtime(utime.mktime(t))
    print(t)
print('gmt')
t=utime.gmtime()
for i in range(5):
    t=utime.gmtime(utime.mktime(t))
    print(t)
```

Thank you!

gijsio, 2020-07-08:
Hi,
You are indeed correct, the timezone gets added twice. I believe this will get fixed in the next revision!
Gijs

gijsio, 2021-01-11:
Hi,
I must admit I was wrong here last time, let me explain,
```
print('utc+timezone = localtime->the following should be 0')
print(utime.time()+utime.timezone() - utime.mktime(utime.localtime()))
```
this will print the timezone setting, as `utime.mktime(...)` is the inverse of `utime.localtime()` (I know, it sounds confusing, but its the consensus in python and C). If you give the localtime as argument, it will output `utime.time()`

Btw, here's my output on 1.20.2.r3:
```
utc+timezone = localtime->the following should be 0
3600
utime.time() is the inverse of utime.gmtime() -> the following is equal
(1970, 1, 1, 0, 10, 25, 3, 1) (1970, 1, 1, 0, 10, 25, 3, 1)
utime.mktime() is the inverse of utime.localtime() -> the following should be equal
(1970, 1, 1, 1, 10, 25, 3, 1) (1970, 1, 1, 1, 10, 25, 3, 1)
This should be 0 but it is twice the timezone
0
This should be -timezone but it is +timezone or twice timezone from the correct value
-3600
The following should be stable values not increasing
local
(1970, 1, 1, 1, 10, 25, 3, 1)
(1970, 1, 1, 1, 10, 25, 3, 1)
(1970, 1, 1, 1, 10, 25, 3, 1)
(1970, 1, 1, 1, 10, 25, 3, 1)
(1970, 1, 1, 1, 10, 25, 3, 1)
gmt
(1969, 12, 31, 23, 10, 25, 2, 365)
(1969, 12, 31, 22, 10, 25, 2, 365)
(1969, 12, 31, 21, 10, 25, 2, 365)
(1969, 12, 31, 20, 10, 25, 2, 365)
(1969, 12, 31, 19, 10, 25, 2, 365)
```

peter-pycom, 2021-01-12:
Not an issue. Closing it.

closed 2021-01-12 by peter-pycom