pycom/pycom-micropython-sigfox #464: IDF hash points to a different commit than the instructions provide

tiborrr, 2020-07-17:
`pycom-micropython-sigfox/esp32$ make`

returns

```
6ec081c is expected from IDF_HASH from Makefile, but
c61fe64 is what IDF_PATH=/mnt/c/Users/tibor/Documents/Programming/pycom-esp-idf is pointing at. 
```

This occurred by following the manual for building from the source. And is because.
`$ git clone --recursive -b idf_v3.3.1 https://github.com/pycom/pycom-esp-idf.git` returns the c61fe64 commit instead of 6ec081c that is expected from the manual.

Either correct the manual or correct makefile.

https://github.com/pycom/pycom-micropython-sigfox/blob/a159dee1ed09c9ab444183da1873c08120db6f1b/esp32/Makefile#L17

peter-pycom, 2020-07-17:
Thanks for flagging this!

6ec081c is correct. 

We shall correct the https://github.com/pycom/pycom-micropython-sigfox/blob/Dev/README.md . When you say "the manual" then this README is what you are referring to, right? Just double checking so I don't miss a place

tiborrr, 2020-07-17:
Yes that is indeed correct

syndycat, 2020-07-22:
I have the same issue. 
6ec081c is expected from IDF_HASH from Makefile, but
2c14666 is what IDF_PATH=/mnt/Programming/pycom/pycom-esp-idf is pointing at.

Do you have a fix?

peter-pycom, 2020-07-22:
well, no, there is no other fix. Checking out this commit and updating the submodules **IS** the fix. The only other thing that needs to be done is correct the README