pycom/pycom-micropython-sigfox #465: TypeError: object with buffer protocol required

rsemple621, 2020-08-03:
(sysname='FiPy', nodename='FiPy', release='1.20.1.r3', version='v1.11-58398d3 on 2019-12-12', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.3.0')

I am just attempting to walk through some of the examples located here @ https://docs.pycom.io/tutorials

I am getting an error when trying to run this section of code in Atom in order to grab the wifi mac address on the pycom:

from network import WLAN
import ubinascii
wl = WLAN()
ubinascii.hexlify(wl.mac())[:6] + 'FFFE' + ubinascii.hexlify(wl.mac())[6:]                <- This line gives the error in the title

Can anyone help?

gijsio, 2020-08-03:
Hi, the `wl.mac()` returns the station and ap mac addresses of the device. You should select either one by `wl.mac().sta_mac` or `wl.mac().ap_mac`. Other than that you are fine!

rsemple621, 2020-08-03:
Appreciate that! I'll use the sta! thanks for the help!

closed 2020-08-03 by rsemple621