pycom/pycom-micropython-sigfox #477: L04 module application firmware upgrade issue.

KiranShingote, 2020-09-07:
Hi Team,

I have designed my own application board and using L04 1.0 module.
Would like to download my application program to L04 module using Wi-Fi communication.
I succeeded in downloading application firmware once, when I received assembled PCB.
Now, I am not able to download the firmware. 
I am doing anything wrong? 

Regards,

Kiran

gijsio, 2020-09-07:
Hi, 
Let me know what firmware version you are on, what firmware version you want to go, what method you are using (FTP, Pybytes), if you registered the Pybytes connection, and what exactly is not working for you (do you get any error messages etc.)

KiranShingote, 2020-09-08:
Hi,

I am using Atom Text Editor & Pymakr Plugin. When I received module, I successfully established connection and downloaded my own application program into it. For the second time, when I tried to connect module, I was not to establish connection. Does module come with factory set program which allowed me to program for the first time? And, now I have overwritten this program and not able to connect. Is it correct?

Regards,

Kiran

robert-hh, 2020-09-08:
If your application program is started in main.py or is main.py, then you will not get a REPL prompt. 
You may try to connect with more simple terminal emulator tools like putty (Windows), screen or picocom(OS X or Linux). There, pushing Ctrl-F or Ctrl-C should stop your application and bring back REPL. Ctrl-F causes a safe boot, which disables execution of boot.py and main.py.

Besides that this place is the bug tracker. Usage questions like yours are better asked in the forum at https://forum.pycom.io, where you will have a broade audience.

KiranShingote, 2020-09-10:
Thanks for suggestion. 
Problem solved after module entering in to safe boot mode.